micropython/micropython #8169: CAN FD not working with STM32 H7

suzonch, 2022-01-13:
I have Nucleo H743ZI, according to https://github.com/micropython/micropython/issues/4774 CAN FD is implemented then why long message isn't working? 
[image]

suzonch, 2022-01-14:
Any suggestion @iabdalkader

iabdalkader, 2022-01-14:
I think I can fix this, but will need you to test it, let me know if you're willing to test it and I'll have a look. Note there's also an open PR to improve and fix some issues with the FDCAN driver.

suzonch, 2022-01-14:
Hi,

I'm totally ready to test it anytime. That'd be really great if it can be fixed. I'm messing around a bit with my limited experience in C.

I'm keeping an eye on that PR too.

Best regards
Md Abu Daud Chowdhury

suzonch, 2022-01-14:
Hi,
I'm totally ready to test it anytime. That'd be really great if it can be fixed. I'm messing around a bit with my limited experience in C.

iabdalkader, 2022-01-16:
I'll let you know when it's done. I got it working, but still need to reconfigure the shared RAM.

suzonch, 2022-01-16:
That's great, so you'll create a new PR with this fix?

iabdalkader, 2022-01-16:
#8185

suzonch, 2022-01-17:
I'll confirm if it works once I get the FD CAN converter working, from logic analyzer it seems to be working(analyzer can't decode CAN FD but only CAN)

dpgeorge, 2022-04-02:
Fixed by ff287d085f95ab0ca3fdf67ee9d8cce6f046b01e

closed 2022-04-02 by dpgeorge