micropython/micropython #8171: MICROPY_OBJ_REPR_D on RISCV64 Icicle Kit port

fcuzzocrea, 2022-01-13:
Hi Micropython community!

In the latest week I worked on porting micropython on a RISCV64 board, namely the [ICICLE Kit from Microchip](https://www.microsemi.com/existing-parts/parts/152514), based on the [Polarfire SoC](https://www.microsemi.com/product-directory/soc-fpgas/5498-polarfire-soc-fpga).
I am using their baremetal library which is available [here](https://github.com/polarfire-soc/platform).

So far, I was able to get a working Micropython port (available [here](https://github.com/fcuzzocrea/micropython/), for who wants to take a look).

However, I can't get working MICROPY_OBJ_REPR_D. Up to mode C, everything works fine, but when I enable the mode D in hal config, the firmware compiles and micropython run, but every operation I do result in TypeError:

```
MicroPython v1.17-330-g803aeed75-dirty on 2022-01-13; mpfs-icicle-kit-es with U51
Type "help()" for more information.
>>> 1+1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 
>>> help()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 
>>> 
```
Does this sound familiar to anyone? Is due to the fact that I am running the port on a 64bit SoC? How can I debug it further?

Thanks!!

dpgeorge, 2022-01-13:
I don't know if it's documented anywhere, but `MICROPY_OBJ_REPR_D` requires a 32-bit architecture.  If the arch is 64-bit then it must be put in 32-bit mode; eg on x86-64 you can force this via the compiler flag `-m32`.

fcuzzocrea, 2022-01-13:
So for a 64bit native machine, there is no way to run micropython with double floating point and long ints ? The reason I want Repr D is to enable representation of full precision floats and integers.  Is there any way to do that on a 64bit machine?

dpgeorge, 2022-01-13:
Just use the default repr A.  Repr D is useful in special cases where you don't want floats to allocate on the heap.

fcuzzocrea, 2022-01-14:
Alright, thanks! But does repr A support double precision floats and 64bit ints ?

dpgeorge, 2022-01-14:
> But does repr A support double precision floats and 64bit ints ?

Double precision floats if you use `#define MICROPY_FLOAT_IMPL (MICROPY_FLOAT_IMPL_DOUBLE)`.  And arbitrary precision ints if you use `#define MICROPY_LONGINT_IMPL        (MICROPY_LONGINT_IMPL_MPZ)`

fcuzzocrea, 2022-01-14:
Alright, thanks for your assistance! :)

closed 2022-01-14 by fcuzzocrea