micropython/micropython #8172: Failing to add WebREPL for the ARDUINO_NANO_RP2040_CONNECT board

huang-lihao, 2022-01-13:
Hi,
I am trying to add WebREPL for the ARDUINO_NANO_RP2040_CONNECT board, but fail.

In **ports/rp2/boards/ARDUINO_NANO_RP2040_CONNECT/mpconfigboard.h**, I add
```c++
// Enable websockets.
#define MICROPY_PY_UWEBSOCKET           (1)
#define MICROPY_PY_OS_DUPTERM           (1)
#define MICROPY_PY_WEBREPL              (1)

// #define MICROPY_PY_LWIP                 (1)
// #define MICROPY_PY_USSL                 (1)
// #define MICROPY_SSL_MBEDTLS             (1)

// use vfs's functions for import stat and builtin open
#define mp_import_stat mp_vfs_import_stat
#define mp_builtin_open mp_vfs_open
#define mp_builtin_open_obj mp_vfs_open_obj
```
and in **ports/rp2/boards/ARDUINO_NANO_RP2040_CONNECT/manifest.py**, I add
```python
include("$(MPY_DIR)/extmod/webrepl/manifest.py")
```

The error message is,
```
[ 56%] Building C object CMakeFiles/firmware.dir/moduos.c.obj
In file included from ***/micropython/py/objstr.h:29,
                 from ***/micropython/ports/rp2/moduos.c:27:
***/micropython/ports/rp2/moduos.c:94:49: error: 'mp_uos_dupterm_obj' undeclared here (not in a function)
   94 |     { MP_ROM_QSTR(MP_QSTR_dupterm), MP_ROM_PTR(&mp_uos_dupterm_obj) },
      |                                                 ^~~~~~~~~~~~~~~~~~
***/micropython/py/obj.h:309:24: note: in definition of macro 'MP_ROM_PTR'
  309 | #define MP_ROM_PTR(p) (p)
      |                        ^
make[3]: *** [CMakeFiles/firmware.dir/build.make:3021: CMakeFiles/firmware.dir/moduos.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:1561: CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [Makefile:84: all] Error 2
make: *** [Makefile:19: all] Error 2
```

I am just a beginner for MicroPython, could anyone help me out? Thank you!

By the way, I am using ubuntu 20.04 on windows wsl.

huang-lihao, 2022-01-14:
I find the solution. In **ports/rp2/moduos.c**, add,
```
#include "extmod/misc.h"
```
Because **mp_uos_dupterm_obj** is defined in **"extmod/misc.h"**.

Should I add a PR doing this? @dpgeorge

iabdalkader, 2022-01-15:
Hi, it's not that easy, in addition to your changes above, the rp2 port is missing SSL support (via mbedtls or axtls), missing `dupterm` support, and `modusocket` is missing some of the needed functions, and it's very unlikely to ever support accept callbacks (so webrepl will have to be run in foreground). That said I did manage to get webrepl working, but not sure if all of the required changes will be accepted.

[image]

huang-lihao, 2022-01-15:
Thank you for your reply!@iabdalkader

Actually, I would like  to build a webserver supporting websocket on `ARDUINO_NANO_RP2040_CONNECT`, which makes data collected by `ARDUINO_NANO_RP2040_CONNECT` able to be shown in a web browser. I think that it is just  a subset of webrepl. I am now working on adding `mbedtls`, but failed.

Could you please send me your codes?

iabdalkader, 2022-01-15:
> Could you please send me your codes?

https://github.com/iabdalkader/micropython/commit/6617cd0779496b764193ff8b83a59131231e5f19

In boot.py or from REPL:
```Python
import sys, uhashlib_sha1
sys.modules['uhashlib'] = uhashlib_sha1
import network, socket, webrepl, webrepl_cfg
SSID=''
KEY=''
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect(SSID, KEY)
webrepl._webrepl.password(webrepl_cfg.PASS)
s = webrepl.setup_conn(8266, None)
s.setblocking(True)
webrepl.accept_conn(s)
```

huang-lihao, 2022-01-16:
It works. Thanks a lot! @iabdalkader

huang-lihao, 2022-01-16:
However, it seems that once `websocket` is disconnected, it won't be connected again.

iabdalkader, 2022-01-16:
> However, it seems that once `websocket` is disconnected, it won't be connected again.

Yes, that's an issue in the firmware because it doesn't have a way to flag the socket addr as reusable, there's no `setsockopt` function/command. Hopefully this will be fixed soon.

TecDroiD, 2022-02-09:
hi,
I compiled the way you mentioned above but haven't got the webrepl import. _webrepl exists.. 
what's needed here?

dpgeorge, 2022-05-26:
WebREPL was enabled on the Nano connect in 875caca504490fb4d95230c06d009502860dc8c3

closed 2022-05-26 by dpgeorge