micropython/micropython #8173: Multiple network interfaces

Tangerino, 2022-01-13:
In a scenario where a platform supports simultaneously:
- 4G - Internet access
- WiFi - Connected to the customer access point for setup/monitoring only (Internet access blocked by proxy)
- Ethernet - Downstream local devices

Or any combination of above options.

Is there a way to tell the socket what interface to use when opening a socket?

Thank in advance

dpgeorge, 2022-01-14:
> Is there a way to tell the socket what interface to use when opening a socket?

I'm not sure.  It will depend a lot on the port and how its network subsystem works.  Which port do you use?

Tangerino, 2022-01-14:
I'm using ESP32.
An in my use case I might need to use the tree interfaces (eth, wlan and ppp), which will be configurable per installation.
If the use case is not clear, please let me know.
Thank you.

Tangerino, 2022-01-17:
I got an Olimex EVB board with Ethernet and was able to test.
All works fine when I try to block each of interfaces to reach the web or by removing the cabe, renaming the AP or blocking its IP address.
The RTOS finds a way to use one or another interface and WiFi is the preferable one.
Thank you and I think this can be closed now.

closed 2022-01-21 by Tangerino