micropython/micropython #8177: Simple example of uasyncio WebSocket Server (secure - with SSL)

beyonlo, 2022-01-14:
Hi all.

I would like a simple example of uasyncio WebSocket Server (secure - with SSL). I see that there is a `uwebsocket` module implemented on MicroPython. 

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.17.0', version='v1.17 on 2021-09-02', machine='ESP32 module (spiram) with ESP32')
>>> import uwebsocket
>>>

```

Well, so I found a example using the `uwebsocket` module: the `webrepl.py`. Is the `webrepl.py` a full example of uasyncio WebSocket Server? 

https://github.com/micropython/micropython/blob/master/extmod/webrepl/webrepl.py

If yes, how is possible to add SSL in that example to secure connections?

Thank you in advance!

Tangerino, 2022-01-16:
You can look here. Not sure about quality, never used.
```url
https://github.com/jczic/MicroWebSrv
```

beyonlo, 2022-01-16:
> You can look here. Not sure about quality, never used.
> 
> ```
> https://github.com/jczic/MicroWebSrv
> ```

@Tangerino 

That use thread, not uasyncio.

bazooka07, 2022-01-19:
Look at __Picoweb__. It uses uasyncio by Paul Sokolovsky.
https://github.com/pfalcon/picoweb
For ssl, I don't know. But it has an example with basic authentication.

beyonlo, 2022-01-19:
Hello @bazooka07 

> Look at **Picoweb**. It uses uasyncio by Paul Sokolovsky. https://github.com/pfalcon/picoweb For ssl, I don't know. But it has an example with basic authentication.

Thanks for the

beyonlo, 2022-01-19:
Hello @bazooka07 

> Look at **Picoweb**. It uses uasyncio by Paul Sokolovsky. https://github.com/pfalcon/picoweb For ssl, I don't know. But it has an example with basic authentication.

The Picoweb do not support websocket. Even uasyncio on PicoWeb is old, it is not the official v3 uasyncio compatible, but it is good start point.

The https://github.com/jczic/MicroWebSrv use WebSocket, but is not official `uwebsocket` module - it wrote their own websocket module. And is better (if is possible) to use the official, for compatibility and so on.

This project  https://github.com/Carglglz/upydev/tree/master/upyutils has a WebREPL example working with SSL using the official uwebsocket module - that is great. This project was put as a PR (https://github.com/micropython/micropython/pull/5611#) to support SSL in the official WebREPL, but I do not know why still not was merged, after two years.

I need just to know now how to add uasyncio on that WebREPL + SSL example.

Carglglz, 2022-12-22:
@beyonlo FYI with the latest changes to #8968 I'm able to run a `microdot_async + tls (https)` server and `WebREPL + tls (wss) + aiorepl`  at the same time, so I think the `uasyncio WebSocket Server (secure - with SSL)` should be possible too. 👍🏼

beyonlo, 2022-12-23:
@Carglglz Excellent!

> so I think the `uasyncio WebSocket Server (secure - with SSL)` should be possible too. 👍🏼

Sorry for my ignorance, but think that already working on the Microdot, like as show this example [echo_async_tls.py](https://github.com/miguelgrinberg/microdot/blob/main/examples/tls/echo_async_tls.py) working with this [index.html](https://github.com/miguelgrinberg/microdot/blob/main/examples/tls/index.html) that use `wss` (`WebSocket Secure Server`). 

Please, let me know what I missing. I'm a bit confuse now: how that (uasyncio WebSocket Server (secure - with SSL)) already working on Microdot if the #8968 is a new implementation to have that working?

Thank you in advance!

Carglglz, 2022-12-24:
@beyonlo I think those examples only work with CPython at the moment?, unless I'm missing something in that example 

```py
import ssl
from microdot_asyncio import Microdot, send_file
from microdot_asyncio_websocket import with_websocket

app = Microdot()

@app.route('/')
def index(request):
    return send_file('index.html')

@app.route('/echo')
@with_websocket
async def echo(request, ws):
    while True:
        data = await ws.receive()
        await ws.send(data)

sslctx = ssl.SSLContext(ssl.PROTOCOL_TLS_SERVER)
sslctx.load_cert_chain('cert.pem', 'key.pem')
app.run(port=4443, debug=True, ssl=sslctx)
```
Right now MicroPython `ssl` module has no `SSLContext`,  and in  `app.run`

This calls 
```py
asyncio.run(self.start_server(host=host, port=port, debug=debug,
                                      ssl=ssl))
```
which calls 
```py
try:
         self.server = await asyncio.start_server(serve, host, port,
                                                     ssl=ssl)
 except TypeError:
            self.server = await asyncio.start_server(serve, host, port)
```
And if you look at  `asyncio.start_server` in MicroPython `extmod/uasyncio/stream.py` there is no `ssl` option 
```py
async def start_server(cb, host, port, backlog=5):
```
Also this comment from @miguelgrinberg in the issue you opened in [#34](https://github.com/miguelgrinberg/microdot/issues/34)
> Would appreciate it if you test it and provide feedback. Note that the async support cannot currently be used with MicroPython, since uasyncio does not have the necessary support

Well, with the latest changes to #8968 the async support is enabled, so those examples will work in MicroPython too. 😄

beyonlo, 2022-12-24:
@Carglglz Ohh, you are right, that [echo_async_tls.py](https://github.com/miguelgrinberg/microdot/blob/main/examples/tls/echo_async_tls.py) do not works with MicroPython. As I still developing using as base example just the [echo_async.py](https://github.com/miguelgrinberg/microdot/blob/main/examples/websocket/echo_async.py)  (WebSocket Server without SSL) and just when the entire application will be stable that I was thinking do add the ssl, that was the reason that I never had tested the [echo_async_tls.py](https://github.com/miguelgrinberg/microdot/blob/main/examples/tls/echo_async_tls.py) on the MicroPython.

> Also this comment from @miguelgrinberg in the issue you opened in [#34](https://github.com/miguelgrinberg/microdot/issues/34)

Yes, I remember now about that moment. Maybe a good idea, as suggstion, is @miguelgrinberg add a note in the that [README.md](https://github.com/miguelgrinberg/microdot/blob/main/examples/tls/README.md) (inside tls examples) talking about that the [echo_async_tls.py](https://github.com/miguelgrinberg/microdot/blob/main/examples/tls/echo_async_tls.py) works (for now :) ) just on the CPython.

> Well, with the latest changes to #8968 the async support is enabled, so those examples will work in MicroPython too. smile

Excellent, so maybe sooner the  [echo_async_tls.py](https://github.com/miguelgrinberg/microdot/blob/main/examples/tls/echo_async_tls.py) will works on MicroPython as well :)

Thank you for the explanation!

dpgeorge, 2023-12-14:
Asyncio now supports SSL server and client mode.  So that's the first and main step to getting secure WebSockets.