micropython/micropython #8189: esp32: PWM fails at low-ish frequencies.

robert-hh, 2022-01-19:
@IhorNehrutsa :
PWM output cannot be set for frequencies lower than about 700 Hz.

Example for a fail:
```
from machine import PWM, Pin
p=PWM(Pin(23, Pin.OUT), freq=100, duty_u16=32768)
```
When starting at 1000Hz, going down in 100Hz steps, the duty cycle changes at 600Hz, and the output is off at 500Hz.

MicroPython v1.18-4-g5e506567a on 2022-01-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.

robert-hh, 2022-01-19:
See: https://forum.micropython.org/viewtopic.php?f=8&t=11836&start=10
The lowest frequency is 611 Hz. That was not the case when I did the first tests with that PR. At that time, PWM was good for frequencies wayl below 1 Hz. But I did not repeat testing at low frequencies later on.

robert-hh, 2022-01-22:
One added observation: if the PWM is initialized with freq<10, it works. Setting freq() to 10 stops the output, setting it to 9 starts it again. So it may be related to the setting of EMPIRIC_FREQ. Setting EMPIRIC_FREQ to 611 opens the range down to 7Hz.
What I see as well is a decreasing frequency resolution at lower frequencies. 10k is fine, below 9980Hz it is of off 0.15%, which is strange, since at lower frequencies the resolution should increase.

robert-hh, 2022-01-22:
Looking into the code, line 314 of machine_pwm looks suspicious.
```
int channel_duty = duty >> (HIGHEST_PWM_RES + UI_RES_SHIFT - timer.duty_resolution);
```
For frequencies < 611, timer.duty_resolution is > 16, and the shift count gets negative.
Changing the line into
```
    int channel_duty;
    if (timer.duty_resolution <= UI_RES_16_BIT) {
       channel_duty = duty >> (HIGHEST_PWM_RES + UI_RES_SHIFT - timer.duty_resolution);
    } else {
       channel_duty = duty << -(HIGHEST_PWM_RES + UI_RES_SHIFT - timer.duty_resolution);
    }
```
fixes the issue, and frequencies down to 1 Hz are fine.

dpgeorge, 2022-02-01:
Fixed by a5e64c209f2d76cb0cc903df875d87e8ca009507

closed 2022-02-01 by dpgeorge