micropython/micropython #8190: ESP32 chips flash size requirements

IhorNehrutsa, 2022-01-19:
It seems obvious, but I doubted...
ESP32 chips are available with different size of SPI flash: from 4Mbit (8,16,32,64) up to 128Mbit:

https://eu.mouser.com/ProductDetail/Espressif-Systems/ESP32-WROOM-32EM113EH3200PS3Q0?qs=sGAEpiMZZMv0NwlthflBix50tZ3s%252BDzjO244H3%2FC%2Fjc%3D

https://eu.mouser.com/ProductDetail/Espressif-Systems/ESP32-WROOM-32DM113DH2800PH3Q0?qs=sGAEpiMZZMv0NwlthflBi0cXBCPPFUUXVsu3VQZ%252B6vg%3D

Latest bin file 
https://micropython.org/resources/firmware/esp32-20220117-v1.18.bin 
has size 1 555 136 bytes == 12 441 088 bits

So ESP32 chips with 4Mb and 8Mb SPI flash sizes are not suitable for MicroPython.
I'm right?

I didn't see this restriction/limitation in the docs.

Please clarify, thanks.
[image]

robert-hh, 2022-01-19:
The two mouser links in your post are for a 4MByte (=32MBits) and a 128MBits (=16MByte) device.

IhorNehrutsa, 2022-01-19:
@robert-hh 
Thanks. 
P.S. And what are we going to do with the Encoder/Counter?

robert-hh, 2022-01-19:
>  And what are we going to do with the Encoder/Counter?

Counting days and Decoding the progress.

closed 2022-01-19 by IhorNehrutsa