micropython/micropython #8193: make: *** mpy-cross: No such file or directory.  Stop.

KJPOUNDY132, 2022-01-19:
Stop. why it says this; make: *** mpy-cross: No such file or directory.  

please help me?

robert-hh, 2022-01-19:
What did you try to do. Building a firmware? In that case mpy-cross should be built as well by the make scripts. You can build it as well manually. Check if the directory micropython/mpy-cross exists. If yes, go into that directory and run `make`.

closed 2022-02-02 by dpgeorge

tolidano, 2024-07-07:
`FileNotFoundError: [Errno 2] No such file or directory: '/home/user/wasp-os/micropython/mpy-cross/mpy-cross'`

there is no mpy-cross folder  INSIDE the mpy-cross folder.

Ran with this command:  `user@ea4532cb73da:~/wasp-os$ make -j `4` BOARD=p8 all`

More output:
```
MPY watch.py
Traceback (most recent call last):
  File "../../tools/makemanifest.py", line 372, in <module>
    main()
  File "../../tools/makemanifest.py", line 305, in main
    res, out = system(
  File "../../tools/makemanifest.py", line 160, in system
    output = subprocess.check_output(cmd, stderr=subprocess.STDOUT)
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/user/wasp-os/micropython/mpy-cross/mpy-cross'
CC gccollect.c
```

tolidano, 2024-07-07:
Once I went into the folder as described above and ran `make BOARD=p8`, then I ran the top level installation make again, and it worked.

TheYigidOS, 2024-07-07:
thx bu ti solvet it!