micropython/micropython #8197: How to use The ESP32 second core?

Tangerino, 2022-01-21:
Is it possible to attach a function to a second core to handle critical conditions?

dpgeorge, 2022-01-21:
See #7258 for a related discussion.  That is now resolved by 357078504d0d2c3a3f06ec4bd69e0ab3bcf9e016, MicroPython now runs on core 1 (the second core).

Tangerino, 2022-01-23:
I'm sorry but did not get your point.

I have a time critical process and wish to run it in the second core so I can handle sec events.
Like _thread.new_thread(core=2, f=func) 

Thank you

mocleiri, 2022-02-03:
You can´t do it in micropython itself but you can configure other FreeRTOS tasks and set the core there.  Then you would communicate between your new tasks and a module in micropython.

For example the machine_i2s implementation has a non-blocking mode where a new FreeRTOS task is created and runs beside micropython and uses a queue to send data back into micropython.

https://github.com/micropython/micropython/blob/fbd47fc46cce80abfc59e680600666473a1868da/ports/esp32/machine_i2s.c#L593

In this case the helper task is running on the same core but that MP_TASK_COREID can be changed to refer to the other core.

jonnor, 2024-09-08:
As described, this can (and must) be done using an added C module.

closed 2024-09-08 by jonnor