micropython/micropython #8202: error v1.18 build esp32 GENERIC_SPIRAM with user modules

erongd, 2022-01-23:
I am trying to build ESP32 GENERIC_SPIRAM board with st7789_mpy native driver.

I build using the v1.18 branch, did it with esp-idf 4.2, 4.2.2 and 4.3.2.

Here is the command I use to build.
make USER_C_MODULES=~/st7789_mpy/st7789/micropython.cmake BOARD=GENERIC_SPIRAM -j 2

Here is the error I am getting with 4.2 and 4.2.2.
Project build complete. To flash, run this command:
/home/erong/.espressif/python_env/idf4.2_py3.7_env/bin/python ../../../esp-idf-4.2/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build-GENERIC_SPIRAM/bootloader/bootloader.bin 0x8000 build-GENERIC_SPIRAM/partition_table/partition-table.bin 0x10000 build-GENERIC_SPIRAM/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    21600  (    7072 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  1584560  (  -11696 remaining)
ERROR: application overflows allocated space of 1572864 bytes by 11696 bytes
make: *** [Makefile:35: all] Error 1

Here is the error I am getting with 4.3.2.
/home/erong/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: micropython.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/home/erong/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/home/erong/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 392 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [Makefile:34: all] Error 2

The same works fine with build for ESP32 GENERIC and ESP32S2 board. And it was working fine with v1.17 branch. Would appreciate any help.

dpgeorge, 2022-01-24:
When building with SPIRAM the compiler needs to add a lot of extra memory instructions to workaround hardware bugs.  So the firmware is much bigger when SPIRAM is enabled.

In v1.18 compilation was changed from -Os to use -O2 to increase performance (on esp32).  Also assertions were disabled.

Two options to fix it are:
1. add the following to sdkconfig.board:
```
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_COMPILER_OPTIMIZATION_PERF=n
```
2. change the factory line in partitions.csv to `factory,  app,  factory, 0x10000, 0x1f0000,`

erongd, 2022-01-24:
I figured the same. Updated the partitions.csv to resolve the problem.

Thanks.

closed 2022-01-24 by erongd

BetterAutomations, 2022-04-08:
> 

Still getting this error even after reverting to 1.17. I wiped out my local micropython files and re-ran git checkout. Tried the sdkconfig options and also tried without them. I don't have a factory line in my partitions file, but my equivalents (ota_0 and ota_1) each are 2MB; I also tried 3MB. (I have an 8MB flash.) Ideas?

BetterAutomations, 2022-04-08:
Found the cause. I had these settings enabled in sdkconfig:
CONFIG_ESP32_WIFI_IRAM_OPT=y
CONFIG_ESP32_WIFI_RX_IRAM_OPT=y
CONFIG_ESP_WIFI_SLP_IRAM_OPT=y