micropython/micropython #8204: ESP32C3 Hardware SPI init error

Walkline80, 2022-01-24:
In file `ports/esp32/machine_hw_spi.c`, [here](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_hw_spi.c#L85) and [here](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_hw_spi.c#L270) defines the dma channel for esp32c3

```c
dma_chan = 1;
```

However, according to [this doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/spi_master.html#_CPPv416spi_common_dma_t), esp32c3 dma channel only support value 0 and 3 (including all IDF versions support for esp32c3)

Here is my test code

```python
from machine import SPI, Pin

spi = SPI(
    1,
    baudrate=40000000,
    polarity=1,
    sck=Pin(6),
    mosi=Pin(7)
)
```

This code will cause the following error:

```docs
spi: spi_bus_initialize(634): invalid dma channel, chip only support spi dma channel auto-alloc
```

My solution is modify the file `ports/esp32/machine_hw_spi.c` [line 271](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_hw_spi.c#L271) to:

```c
        #if CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3
```

Eivs, 2022-03-05:
I have the same problem

`MicroPython v1.18 on 2022-01-17; ESP32C3 module with ESP32C3
Type "help()" for more information.
>>> from machine import Pin, SPI
>>> hw_spi = SPI(1,  sck=Pin(2),  mosi=Pin(2),  miso=Pin(10))
`

`
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: invalid configuration
`

robert-hh, 2022-03-05:
That seems to be a different error: you assigned the same pin to sck and mosi.

Eivs, 2022-03-05:
Thanks for the reply

I still get the same result after changing to the correct pin

[image]

```
>>> hw_spi = SPI(1,  sck=Pin(2),  mosi=Pin(3),  miso=Pin(10))
  Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
  OSError: invalid configuration
```

laxtiz, 2022-03-06:
i have some board from luatos's shop, and flash that firmware [esp32c3](https://micropython.org/download/esp32c3/).

[image]

Eivs, 2022-03-07:
> i have some board from luatos's shop, and flash that firmware [esp32c3](https://micropython.org/download/esp32c3/).
> 
> [image]

HI！are you in WH?  I'm in Optics Valley .

Walkline80, 2022-03-08:
@Eivs @laxtiz maybe you could setup debug level to `LOG_WARN` for more details ([esp.osdebug](https://docs.micropython.org/en/latest/library/esp.html?highlight=esp%20osdebug#esp.osdebug))

Eivs, 2022-03-08:
Thanks, I got the same error

E (375164) spi: spi_bus_initialize(762): invalid dma channel, chip only support spi dma channel auto-alloc

robert-hh, 2022-03-08:
I can make a PR using the code suggested by @Walkline80, unless it's done by her/him.

Eivs, 2022-03-09:
On my board, test is passed，But I don't know why it only works at 40MHZ

robert-hh, 2022-03-09:
40MHz COU freq or 40 MHz SPI clk?

robert-hh, 2022-03-09:
I just repeated my test with a Node-MCU esp32-c3-32S-kit. SPI works fine at a CPU freq of 160 MHz and given baud rate. Test lines:
```
from machine import SPI, Pin
spi=SPI(1, sck=Pin(8), mosi=Pin(2), miso=Pin(3), baudrate=1_000_000)
spi.write(b'hello world')
b=bytearray(11)
spi.write_readinto(b'hello world', b)
```
Same with 10 MHz or 100kHz

dpgeorge, 2022-03-21:
Fixed by 09b55dc297da2b4dc3915f2a4cc0e3b53ea07359

closed 2022-03-21 by dpgeorge