micropython/micropython #8208: RP2: Overclock or CPU_MHZ as a board config option

Gadgetoid, 2022-01-25:
For PicoSystem we shipped a port of MicroPython that clocks the RP2040 to 250MHz at 1.2v (up from the stock 133MHz 1.1v settings) and have (short of needing to apply the slight overvolt in response to some units not being stable) generally seen these settings to be stable and safe. I accomplished this by maintaining a fork full of hacks where I've modified `main.c` :grimacing: 

Many RP2040-based projects leveraging MicroPython could probably benefit from this same overclock, particularly if the manufacturers are happy to support boards at these speeds.

As such I'd like to propose a board config approach to setting CPU frequency and overvolt, with a view to supporting this in the RP2 port at first and perhaps expanding to other boards (we've encountered MCUs that are ostensibly one speed, but get a manufacturer certified OC frequency later).

This is part of a multi-step process to untie all the hacks I needed to implement for PicoSystem, justify their existence and bring them upstream for the benefit of everyone.

It also seems a shame to leave that 2x+ speedup on the table.

1. What settings are common across MCUs for achieving overclock? I guess CPU_FREQ and VOLTAGE perhaps, but the latter could be derived from the former in most cases
2. Is there anywhere in the MicroPython codebase where assumptions are made about CPU frequency that this would horribly break?
3. Would it be appropriate to configure an "Overclocked Pico" board variant, for those who want to opt in to more performance?

Gadgetoid, 2022-01-26:
Looks like this is tangentially related: https://github.com/micropython/micropython/pull/6941

I think there's still a good case for an overclock-by-default option on a board by board basis.

dpgeorge, 2022-02-02:
> This is part of a multi-step process to untie all the hacks I needed to implement for PicoSystem, justify their existence and bring them upstream for the benefit of everyone.

A great goal!

> As such I'd like to propose a board config approach to setting CPU frequency and overvolt, with a view to supporting this in the RP2 port at first and perhaps expanding to other boards

This sounds fine to me.

There is the existing `machine.freq()` function which can be used to change the CPU frequency, and that allows overclocking on ports/MCUs that support it (eg stm32, rp2).  So the user can dynamically opt-in to overclocking.

> 1. What settings are common across MCUs for achieving overclock?

I think this is very hardware dependent.  Apart from providing a common `machine.freq()` function (which already exists) I don't think it's worth trying to define a common way to build firmware with overclocking enabled.

> 2\. Is there anywhere in the MicroPython codebase where assumptions are made about CPU frequency that this would horribly break?

No.

> 3\. Would it be appropriate to configure an "Overclocked Pico" board variant, for those who want to opt in to more performance?

I don't think so.  Users can just do `machine.freq(250_000_000)` to overclock when they need to, any any existing board.

> I think there's still a good case for an overclock-by-default option on a board by board basis.

Yes I agree.  Would be good to have a build option (for rp2) to select the default clock freq.  Feel free to make a PR!