micropython/micropython #8209: Frozen MPY in fodler are overrided by folder on VFS

petrkr, 2022-01-25:
in 1.18 (exact commit do not know) was proabbly changed behavior with FROZEN modules.

Consider situation you frozed few libraries in custom build under some path, like
FROZEN: /utils/foo.mpy

when you go to REPL, you can use it by 
import utils.foo

But when you create folder on VFS /utils and upload another library bar.py (or bar.mpy), then foo stop works
```
>>> import utils.foo
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'utils.foo
```

in some older revisions this works fine and prioritized search in frozen, then /if not found/ used VFS path.

if '.frozen' and '' are swapped in sys.path, then frozen modules start works, but VFS stop works

dpgeorge, 2022-01-27:
Yes, support for `".frozen"` in `sys.path` was added in e0bf4611c3a8b23b3c52e6a7804aac341ac3a87d, see #8079 and all the linked issues to that for discussion and background.

It looks like what you were previously doing -- which no longer works -- is having a package (in the above case `utils`) split over multiple directories.  As far as I know this is not supported in standard CPython, and the fact it did previously work in MicroPython was an unintended feature....!

I would suggest naming the packages by separate names, those that are on the VFS and those that are frozen.  Eg `utils` and `frozen_utils`.  Then they will not clash and it will always be clear where some code came from (either the VFS or frozen).

petrkr, 2022-01-27:
I see..

I do not know about frozen modules in CPython (if it's normal python in linux). So I do not know it's behavior.

In uPy I using it like pre-compiled utilities and user can add more if wants them. If I will use some other prefix (lets say frozen_utils) it will lead to user confuse...

But let's do example for example on drivers

you will have frozen driver for some LCD in folder "drivers", so you have driver.oled. But if user want add more driver, they just upload another to VFS driver folder and use them too. But in some usage library you can do import drivers.* and then select or whatever.

For now I can do buch of workarounds, but for example now I am using some GUI (microgui) and there are lot of widgets, but if I will want freeze (memory, ram etc) some core to firmware, I can not add more of them to VFS too, because same directory.

For now I am using only early-libraries, which I pre-compile to my image and they will download rest of framework from TAR from web server.. So I can duplicate those libraries to TAR, so frozen will NOT be used anymore, but it's kinda waste to have them twice, but on other side it gives me way how to update even frozen modules by upload newer version to VFS...