micropython/micropython #8211: ESP32 build fails building i2s

dcmcshan, 2022-01-26:
I am building ESP32 port as per instructions and run across this when building...

/Users/danielmcshan/GitHub/micropython/ports/esp32/machine_i2s.c: In function 'machine_i2s_init_helper':
/Users/danielmcshan/GitHub/micropython/ports/esp32/machine_i2s.c:471:15: error: 'i2s_config_t' {aka 'struct <anonymous>'} has no member named 'mclk_multiple'
     i2s_config.mclk_multiple = I2S_MCLK_MULTIPLE_DEFAULT;
               ^
[ 98%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/modnetwork.c.obj
/Users/danielmcshan/GitHub/micropython/ports/esp32/machine_i2s.c:471:32: error: 'I2S_MCLK_MULTIPLE_DEFAULT' undeclared (first use in this function); did you mean 'RTC_CLK_CONFIG_DEFAULT'?
     i2s_config.mclk_multiple = I2S_MCLK_MULTIPLE_DEFAULT;
                                ^~~~~~~~~~~~~~~~~~~~~~~~~
                                RTC_CLK_CONFIG_DEFAULT
/Users/danielmcshan/GitHub/micropython/ports/esp32/machine_i2s.c:471:32: note: each undeclared identifier is reported only once for each function it appears in
/Users/danielmcshan/GitHub/micropython/ports/esp32/machine_i2s.c:472:16: error: 'i2s_config_t' {aka 'struct <anonymous>'} has no member named 'bits_per_chan'; did you mean 'bits_per_sample'?
     i2s_config.bits_per_chan = 0;
                ^~~~~~~~~~~~~
                bits_per_sample
/Users/danielmcshan/GitHub/micropython/ports/esp32/machine_i2s.c:491:16: error: 'i2s_pin_config_t' {aka 'struct <anonymous>'} has no member named 'mck_io_num'; did you mean 'bck_io_num'?
     pin_config.mck_io_num = I2S_PIN_NO_CHANGE;
                ^~~~~~~~~~
                bck_io_num
make[3]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/__/machine_i2s.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make[1]: *** [all] Error 2

dpgeorge, 2022-01-27:
What version of the IDF are you using?  Please make sure to use a supported one (see the esp32 README).

miketeachman, 2022-01-27:
This is failing on the new structure members that were added in IDF 4.4.  The recent PR(https://github.com/micropython/micropython/pull/8110) was intended to conditionally compile these new members only for IDF 4.4 and above.  To debug this issue further it will be important to know the exact IDF version you are using.  Thanks.

dcmcshan, 2022-02-02:
Yeah, I was using a custom port to Apple Silicon M1. Thanks for the explanation.

jonnor, 2024-07-07:
This should not be a problem if using a supported esp-idf version.

closed 2024-07-07 by mattytrentini