micropython/micropython #8212: SAMD I2C Module

andrewteta, 2022-01-27:
I have a need for I2C and would like to use MicroPython with the Adafruit Feather M4 Express (ATSAMD51J19). Since support for the SAMD family is lacking, I've started by defining a new board definition in my fork [andrewteta/micropython](https://github.com/andrewteta/micropython). Next, I'd like to begin implementing the ASF4 I2C drivers for the SAMD51.

Briefly looking at the documentation there, I have a few options. My plan right now is to use the I2C master synchronous driver.

This is my first time attempting to develop for this project, though I've been following and reading docs for quite some time. This means I will likely need some guidance with the workflow if/when it comes time to incorporate any of my code into the upstream repo.

dhalbert, 2022-01-27:
https://github.com/adafruit/circuitpython uses ASF4. You could look at our implementations as a start.

ASF4 was a headache. We had to make up a "typical and complete" project in start.atmel.com, then download that, and then put that in a repo (the submodule that both CIrcuitPython use). There is no downloadable zip file of all of ASF4 that is easily kept up to date. ASF3 was much more straightforward.

ASF4 is also expecting fixed pin assignments, and we had to do trickery to get around that.

We started a very thin HAL library for SAMD, but it is very incomplete: https://github.com/adafruit/samd-peripherals.

andrewteta, 2022-01-28:
@dhalbert Thanks for pointing me there! After spending a few long hours diving into it last night I came to all the same conclusions. My next step will be to go through the CircuitPython stuff. The ASF4 project configuration looks to be set up correctly, but I failed to get past an i2c init call. It’s too bad the ASF4 example code is so minimal.

dhalbert, 2022-01-28:
@adnrewteta We use the ASF4 configurations as a "prototypical" source of settings, and then use it dynamically on the pins in questions. That is, in the START project, we defined I2C on one set of pins as an example, and then we reuse those settings for whatever pins the user has specified. (The original pins are not tied to those settings.)

closed 2022-12-15 by robert-hh