micropython/micropython #8214: error in documentation

labodezao, 2022-01-27:
In here :

https://github.com/micropython/micropython/wiki/Programming-Debugging-the-pyboard-using-ST-Link-v2

On pyboard side : 
 pyboard v1.1
----------------- | -------------
PIN1  (VAPP)      -> 3V3
PIN7  (TMS_SWDIO) -> PA13 (X21-2)
PIN9  (TCK_SWCLK) -> PA14 (X20-2)
PIN13 (TDO_SWO)   -> PB3  (X17-1) (Optional debug serial)
PIN15 (NRST)      -> RST
PIN20 (GND)       -> GND

PA13 is P5 and Pa14 P4 not X21 and X20

jonnor, 2024-09-08:
Thank you for reporting. The documentation is found in the docs/ directory, and contributions in form of merge requests are welcomed.

andypiper, 2024-09-10:
This is in the wiki, which I have access to update - I don't have a Pyboard handy to confirm this but I can go ahead and update to close this out.

andypiper, 2024-09-10:
fixed.

closed 2024-09-11 by dpgeorge