micropython/micropython #8217: Failure To Mount SD Card

TGiles1998, 2022-01-27:
ESP32 S3 SD Card class not working.

https://github.com/micropython/micropython/issues/8121

TGiles1998, 2022-01-27:
Trying to use the pins alsop used for octal RAM to wire up the SD card from ports/esp32/machine_hw_spi.c

// ESP32S2 and S3 uses GPIO matrix for SPI3 pins, no IO_MUX possible
// Set defaults to the pins used by SPI2 in Octal mode
#define MICROPY_HW_SPI2_SCK (36)
#define MICROPY_HW_SPI2_MOSI (35)
#define MICROPY_HW_SPI2_MISO (37)

sck = machine.Pin(36)
miso = machine.Pin(37)
mosi = machine.Pin(35)
cs = machine.Pin(10)
sd = machine.SDCard(slot=1, sck=sck, miso=miso, mosi=mosi, cs=cs)
os.mount(sd, '/sd')

returns error:

E (59450) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 16

TGiles1998, 2022-01-27:
{
        // SPI interface
        #if CONFIG_IDF_TARGET_ESP32S3
        STATIC const sdspi_slot_config_t slot_defaults[2] = {
            {
                .gpio_miso = GPIO_NUM_36,
                .gpio_mosi = GPIO_NUM_35,
                .gpio_sck = GPIO_NUM_37,
                .gpio_cs = GPIO_NUM_34,
                .gpio_cd = SDSPI_SLOT_NO_CD,
                .gpio_wp = SDSPI_SLOT_NO_WP,
                .dma_channel = 2
            },
            SDSPI_SLOT_CONFIG_DEFAULT()

The GPIO 34 for chip select is not accessible on the devkitc board.

TGiles1998, 2022-01-28:
Changing:

{
// SPI interface
#if CONFIG_IDF_TARGET_ESP32S3
STATIC const sdspi_slot_config_t slot_defaults[2] = {
{
.gpio_miso = GPIO_NUM_36,
.gpio_mosi = GPIO_NUM_35,
.gpio_sck = GPIO_NUM_37,
.gpio_cs = GPIO_NUM_34,
.gpio_cd = SDSPI_SLOT_NO_CD,
.gpio_wp = SDSPI_SLOT_NO_WP,
.dma_channel = 2
},
SDSPI_SLOT_CONFIG_DEFAULT()

to:

 #if CONFIG_IDF_TARGET_ESP32S3
        STATIC const sdspi_slot_config_t slot_defaults[2] = {
            {
                .gpio_miso = GPIO_NUM_13,
                .gpio_mosi = GPIO_NUM_11,
                .gpio_sck = GPIO_NUM_12,
                .gpio_cs = GPIO_NUM_10,
                .gpio_cd = SDSPI_SLOT_NO_CD,
                .gpio_wp = SDSPI_SLOT_NO_WP,
                .dma_channel = 2
            },
            SDSPI_SLOT_CONFIG_DEFAULT()
        };

Error Produced:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 16

TGiles1998, 2022-01-28:
https://github.com/espressif/esp-idf/issues/8257

TGiles1998, 2022-01-29:
Used 10k Pull up resistors on the data lines and still failed. Will look into ESP-IDF side of the build.

TGiles1998, 2022-03-15:
Solved: create a timeout and continue to os.mount() until connects

closed 2022-03-15 by TGiles1998

ChihoSin, 2022-04-18:
> Solved: create a timeout and continue to os.mount() until connects

sck = machine.Pin(36)
miso = machine.Pin(37)
mosi = machine.Pin(35)
cs = machine.Pin(10)
sd = machine.SDCard(slot=1, sck=sck, miso=miso, mosi=mosi, cs=cs)
time.sleep(2)
sd.info()

>> E (171500) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107

Same error with timeout before read.