micropython/micropython #8218: ESP32 Port - Using UART0

Tangerino, 2022-01-28:
Any particular reason the UART0 is blocked for user code?
What are the implication of using it? What are the `fun errors`?
This can drive the hardware design choices.
Thank you

```c
STATIC mp_obj_t machine_uart_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *args) {
    mp_arg_check_num(n_args, n_kw, 1, MP_OBJ_FUN_ARGS_MAX, true);

    // get uart id
    mp_int_t uart_num = mp_obj_get_int(args[0]);
    if (uart_num < 0 || uart_num >= UART_NUM_MAX) {
        mp_raise_msg_varg(&mp_type_ValueError, MP_ERROR_TEXT("UART(%d) does not exist"), uart_num);
    }

    // Attempts to use UART0 from Python has resulted in all sorts of fun errors.
    // FIXME: UART0 is disabled for now.
    if (uart_num == UART_NUM_0) {
        mp_raise_msg_varg(&mp_type_ValueError, MP_ERROR_TEXT("UART(%d) is disabled (dedicated to REPL)"), uart_num);
    }
```

UnexpectedMaker, 2022-01-29:
If you have a Serial2Uart chip on your board, then UART0 is locked out by it anyway.  ESP32 is not Native USB. It uses external hardware and that hardware controls the UART0 pins, so they are not free to use by the user.

Tangerino, 2022-01-29:
Well, in my design, I can route the pins to another destination  and decide, after the user code runs, to execute other function like a regular serial communication.
The question is what happen if I use this serial it will work?

UnexpectedMaker, 2022-01-29:
> Well, in my design, I can route the pins to another destination and decide, after the user code runs, to execute other function like a regular serial communication.

Sure, but you'd then have the only ESP32 board on the market that doesn't conform with UART0 on the USB via a Serial2UARt chip. Your shoe if you want to go against 1000 other board designs, and how Arduino, MP and IDF are all setup.

> The question is what happen if I use this serial it will work?

Not without making custom MP firmware specific to your board & design.

UnexpectedMaker, 2022-01-29:
The question is, why do you want to make this change that will make your board incompatible with everything else? I can't possibly think of a good reason why. Those pins (UART0) are pretty useless as GPIO for anything else. 

They are not touch, or ADC or RTCIO...  so why not just leave them as UART0?

Tangerino, 2022-01-29:
This is my design, and the board is attached to USB as any other board.
I can, via jumpers, route these pins to a RS485 transceiver.
Based on another jumper (RUN/DEBUG), I tell the user code to stop REPL and use UART0 to communicate to external devices.
But the point here is just the question on the MP code that does not let us to use this UART0.
The original question is Why?

IhorNehrutsa, 2022-01-31:
https://forum.micropython.org/viewtopic.php?f=15&t=11436&p=62614#p62567

beyonlo, 2022-01-31:
I agree with @Tangerino 

I have a design where I would like to use that UART 0 pins to use as RS485. But when I need to program the firmware via esptool, that UART 0 will works. So, that UART 0 pins will be in parallel with 485.

Anyway @UnexpectedMaker as you told about that is a non native USB, could you tell me if in the S3 USB (that is used to program), the Micropython give the user the permission to use that pins to for another porpose too? I mean, I would like to use the USB (that two pins) for ADC, but when I will program a new firmware is possible to program using the USB too. So in hardware design will have parallel use that two pins of USB. Is possible?

Thank you in advance!

jonnor, 2024-09-08:
UART 0 being hard set as USB/REPL seems to be as-designed for MicroPython on ESP32.

@beyonlo  On ESP32-S3 it should be possible to use UART 0 for other functions.

projectgus, 2024-09-10:
I think this is a reasonable feature request, although it's something of a niche use case. The `FIXME` in the code suggests the original implementer thought so as well!

The current implementation is quite different, but it still doesn't work. i.e.:

```
>>> import machine
>>> u0 = machine.UART(0)
E (13243) uart: uart_wait_tx_done(1231): uart driver error
E (13243) uart: call uart_driver_install API first
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
```

> @beyonlo On ESP32-S3 it should be possible to use UART 0 for other functions.

I think you still need to do a custom build for this, as `MICROPY_HW_ENABLE_UART_REPL` is generally set. If it's set to 0, `machine.UART(0)`.

---

I haven't looked closely, but I think the difficult part here from the MicroPython perspective is ensuring that MicroPython stdin/stdout is fully disabled on UART0 in this situation - because otherwise various unwanted behaviours will show up, as per the original comment. So most likely this will stay open until/unless someone decides to put the work in to implement and check this part.

(There have been discussions on making all stdin/stdout more Python-based (i.e. so that MicroPython stdin/stdout actually uses the `machine.UART(0)` object, for example. However this is an even higher bar than what's requested here!)

Aleksandr66, 2025-01-13:
Why is it so difficult to turn it off? What would free up UART0 for Async StreamReader?
The buffer cannot return less than specified in the byte =(
will come in handy to read 1 at a time