micropython/micropython #8223: Q/FR: Arduino RP2400 Atecc support

mattmeye, 2022-01-29:
Hello, First let me thank you for your amazing project. 

I have a question, is a support for Atecc chip in the Arduino rp2400 planned?

I would like to create and save certs for iot communication safely.

Kind regards, matt

dpgeorge, 2022-01-29:
@iabdalkader do you have any plans to implement a driver for the ATECC608A?

iabdalkader, 2022-01-29:
There's already a library for micropython that looks very promising: https://github.com/dmazzella/ucryptoauthlib. As you can see it's a lot of work, so it doesn't make sense to rewrite it. If it's add as a submodule, somewhere, we can just include it in the board manifest.

mattmeye, 2022-01-30:
Sounds good, I would appreciate that!

mattmeye, 2022-02-06:
I have no experience with Micropython, but could try to support with a few hints. Is there already a library that has been integrated via submodule?

iabdalkader, 2022-02-10:
> Is there already a library that has been integrated via submodule?

No.