micropython/micropython #8224: utime.localtime() shows wrong day of week

jorivaa, 2022-01-29:
_I am getting wrong days of week with utime.localtime() on Raspberry Pico._

_Steps to reproduce bug:_

import machine
tm=(2022,1,29,0,11,01,15,0) ### ( y,m,d, 0 , h,m,s, 0 )
machine.RTC().datetime(tm)
import utime
utime.localtime()

_This shows:_
(2022, 1, 29, 11, 1, 24, 4, 29)

_The seventh field shows "4" which should be 5.
So the day-of-week is off by one._

_This is my micropython version:_
import os
os.uname()
(sysname='rp2', nodename='rp2', release='1.18.0', version='v1.18 on 2022-01-17 (GNU 11.2.0 MinSizeRel)', machine='Raspberry Pi Pico with RP2040')

peterhinch, 2022-01-29:
See [this forum thread](https://forum.micropython.org/viewtopic.php?f=21&t=11108&p=61038).

dpgeorge, 2022-01-29:
Thanks for the report.  But I'll close as a duplicate of #7889.

closed 2022-01-29 by dpgeorge