micropython/micropython #8232: ESP32 and Zephyr: Error in nlr_push_tail call

sylvioalves, 2022-01-31:
Hi, I have tried testing micropython porting for zephyr and ESP32 board. Got myself into this error:  
```c
*** Booting Zephyr OS build v3.0.0-rc1-34-gfe4ab778886c  ***
MicroPython v1.18-63-g60e05ae84 on 2022-01-31; zephyr-esp32 with esp32
>>> print("hello")

Backtrace:0x400E1A39:0x3FFE64E0 
0x400e1a39: nlr_push_tail at /home/sylvio/projects/espressif/zephyr/micropython/py/nlr.c:42
```
Steps to reproduce:
`rm -rf build; west build -b esp32 -- -DCONF_FILE=prj_minimal.conf`
`west flash; west espressif monitor`

dpgeorge, 2022-02-02:
I'm unable to compile zephyr for the esp32, I get errors like this (using zephyr 3.0.0-rc1):
```
In file included from zephyr/lib/posix/pthread_common.c:10:
zephyr/include/posix/time.h:90:15: error: static declaration of 'clock_gettime' follows non-static declaration
 __syscall int clock_gettime(clockid_t clock_id, struct timespec *ts);
               ^~~~~~~~~~~~~
In file included from zephyr/include/posix/time.h:12,
                 from zephyr/lib/posix/pthread_common.c:10:
.espressif/tools/zephyr/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/time.h:187:5: note: previous declaration of 'clock_gettime' was here
 int clock_gettime (clockid_t clock_id, struct timespec *tp);
     ^~~~~~~~~~~~~
In file included from zephyr/lib/posix/pthread_common.c:10:
zephyr/include/posix/time.h:94:5: error: conflicting types for 'timer_create'
 int timer_create(clockid_t clockId, struct sigevent *evp, timer_t *timerid);
     ^~~~~~~~~~~~
In file included from zephyr/include/posix/time.h:12,
                 from zephyr/lib/posix/pthread_common.c:10:
.espressif/tools/zephyr/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/time.h:192:5: note: previous declaration of 'timer_create' was here
 int timer_create (clockid_t clock_id,
     ^~~~~~~~~~~~
```
How did you manage to build it, what steps did you take?

sylvioalves, 2022-05-27:
@dpgeorge Hi, would you try again using latest 3.1.0-rc2? Those issues you mentioned above are fixed already.

c4explosive, 2022-07-16:
With version 3.1.99 i have the same issue as @dpgeorge . (Don't compiles because of conflicting headers files (time.h) between zephyr and espressif).

c4explosive, 2022-07-16:
> With version 3.1.99 i have the same issue as @dpgeorge . (Don't compiles because of conflicting headers files (time.h) between zephyr and espressif).

Ok, i could successfully compile. With the latest version of Zephyr, all of these variables:

- ZEPHYR_TOOLCHAIN_VARIANT
- ESPRESSIF_TOOLCHAIN_PATH

**Should NOT be set.** Zephyr have it's own version of compiler and toolchain for esp32, so previous espressif's toolchain is no longer needed installed.

bertouttier, 2024-02-09:
@sylvioalves were you able to fix your original problem on esp32s3?