micropython/micropython #8233: STM32L476RG - micropython v1.18 - not a valid ADC channel: 1..2..3

ledtecgm, 2022-01-31:
In the firmware version v1.18 and the port stm32 (STM32L476RG),  I cannot assign an ADC function to the corresponding ports, such as PC0, PC1, PC2 and so on. In the firmware version v1.17 everything still works as desired. Can it be that it is related to the change to STM32lib HAL v1.17.0?

iabdalkader, 2022-01-31:
There were changes to the ADC driver and ADC pins introduced recently, but I also noticed the L4 channels are not mapped from decimal numbers to ADC channels. Does this PR #8235 fix the issue ?

dpgeorge, 2022-02-01:
Can you please provide code snippets that no longer work on V1.18?

ledtecgm, 2022-02-01:
Hello, yes of course, ich can provide some init code for der ADC function.
Many thanks for the quick reply.
I will now continue to use  the FW v1.17 again, since all my required hardware modules work there.

**firmware:** v1.17
MicroPython v1.17 on 2021-09-02; NUCLEO-L476RG with STM32L476RG

Type "help()" for more information.
>>> adc123_in1 = pyb.ADC(pyb.Pin.cpu.C0)
>>> adc123_in1
<ADC on Pin(Pin.cpu.C0, mode=Pin.ANALOG) channel=1>
>>> 

**firmware:** v1.18
MicroPython v1.18 on 2022-01-17; NUCLEO-L476RG with STM32L476RG

Type "help()" for more information.
>>> adc123_in1 = pyb.ADC(pyb.Pin.cpu.C0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: not a valid ADC Channel: 1
>>>

ledtecgm, 2022-02-01:
> There were changes to the ADC driver and ADC pins introduced recently, but I also noticed the L4 channels are not mapped from decimal numbers to ADC channels. Does this PR #8235 fix the issue ?

Sorry, I don't know enough about this to be able to answer the question.

dpgeorge, 2022-02-03:
Fixed by e5df4a96fa85ad4cf10d82b3cd5f8ca8b279050b

closed 2022-02-03 by dpgeorge

ledtecgm, 2022-02-04:
Hello everyone, I have checked the firmware of the "nightly build _v1.18-89-g5679fe6ae (2022-02-04)_", now the ADC assignment works as desired, many thanks for the quick support. Kind regards Guido