micropython/micropython #8234: v1.11 compilation error

jmvilla0, 2022-01-31:
Hi, I'm trying to compile the 1.11 version but I always get errors, if I compile the 1.18 version everything is alright. But I need this old version, the documentation into the README of the 1.11 version is not so clear

> Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path:       /home/jose/esp/esp-idf
Current git hash:   c40f2590bf759ff60ef122afa79b4ec04e7633d2
Supported git hash (v3.3): 6ccb4cf5b7d1fdddb8c2492f9cbc926abaf230df
Supported git hash (v4.0-beta1) (experimental): 310beae373446ceb9a4ad9b36b5428d7fdf2705f
GEN build-GENERIC/sdkconfig.h
Traceback (most recent call last):
  File "/home/jose/esp/esp-idf/tools/kconfig_new/confgen.py", line 641, in <module>
    main()
  File "/home/jose/esp/esp-idf/tools/kconfig_new/confgen.py", line 261, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "/home/jose/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 947, in __init__
    self._init(filename, warn, warn_to_stderr, encoding)
  File "/home/jose/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1085, in _init
    self._parse_block(None, self.top_node, self.top_node).next = None
  File "/home/jose/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 2970, in _parse_block
    raise KconfigError(
kconfiglib.KconfigError: /home/jose/esp/esp-idf/Kconfig:165: '$COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE' not found (in 'source "$COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
make: *** [Makefile:335: build-GENERIC/sdkconfig.h] Error 1

I've already tried to use different ESP-IDF versions but nothing is working

fivdi, 2022-01-31:
If uplatform is available in v1.11 (I'm not sure if it is), then running the following in the repl may reveal which version of esp-idf was used at compile time:
```py
import uplatform
uplatform.platform()
```
As an example, here is what I see with v1.18 on an ESP32:
```
MPY: soft reboot
MicroPython v1.18 on 2022-01-17; ESP32 module with ESP32
Type "help()" for more information.
>>> import uplatform
>>> uplatform.platform()
'MicroPython-1.18.0-xtensa-IDFv4.2.2-with-newlib3.0.0'
```
Alternatively, on Linux, the strings command can be used to extract text from binary files. grep can then be used to search the text for "idf" like this:
```
$ strings esp32-20220117-v1.18.bin | grep idf
/home/micropython/esp-idf-v4.2/components/freertos/tasks.c
/home/micropython/esp-idf-v4.2/components/freertos/timers.c
/home/micropython/esp-idf-v4.2/components/newlib/locks.c
idf_heap_info
/home/micropython/esp-idf-v4.2/components/driver/adc_common.c
/home/micropython/esp-idf-v4.2/components/driver/dac_common.c
/home/micropython/esp-idf-v4.2/components/driver/i2c.c
/home/micropython/esp-idf-v4.2/components/driver/i2s.c
/home/micropython/esp-idf-v4.2/components/driver/rtc_io.c
/home/micropython/esp-idf-v4.2/components/driver/touch_sensor_common.c
/home/micropython/esp-idf-v4.2/components/driver/esp32/touch_sensor.c
/home/micropython/esp-idf-v4.2/components/freertos/event_groups.c
Invalid coexist adapter function md5, internal: %s, idf: %s
Invalid OSI esp_wifi_os_adapter.h md5, internal: %s, idf: %s
Invalid WiFi esp_wifi_crypto_types.h md5, internal: %s, idf: %s
Invalid WiFi esp_wifi_types.h md5, internal: %s, idf: %s
Invalid WiFi esp_wifi.h md5, internal: %s, idf: %s
Invalid supplicant md5, internal: %s, idf: %s
/home/micropython/esp-idf-v4.2/components/esp_ringbuf/ringbuf.c
/home/micropython/esp-idf-v4.2/components/freertos/queue.c
$ 
```

dpgeorge, 2022-02-01:
You need to check out the ESP-IDF at commit 6ccb4cf5b7d1fdddb8c2492f9cbc926abaf230df and update the submodules.  Also make sure the esp32 toolchain is the right version.  Then it should build.

closed 2022-02-01 by dpgeorge

jmvilla0, 2022-02-01:
It works, thanks <3

kindmartin, 2022-05-13:
@dpgeorge  Im trying to compile for a ESP32 port this 1,9.4 release and doing so getting some IDF error. ref as https://github.com/micropython/micropython/issues/8607.

What would be the right ISF version to use to compile this port?