micropython/micropython #8241: Frozen python code as part of machine module?

harbaum, 2022-02-01:
I am implementing an alternative to https://github.com/micropython/micropython/pull/6639 which only does the very basic ESP32 PCNT API in C. The Counter and Encoder classes as decribed at https://github.com/micropython/micropython/pull/8072 are pure Python and subclassed from this PCNT API.

I personally think that such an approach is nicer and cleaner than the current approach. I thus would like to provide a full implementation of my approach.

But Encoder and Counter classes a per the aforementioned documentation part of the machine module. And I don't know how to add Python code to that. Is that possible at all? Can I add python code to the machine module to be frozen into the firmware?

I simply tried to add machine/Encoder.py to port/esp32/modules. But that does not work. It compiles but Encoder does not how up in the machine module.

dpgeorge, 2022-02-01:
There is not a direct way to do this, to combine Python and C code in the same (built in) module.

But the idea with u-module naming is that the C parts can be implemented in, eg, `umachine` and then the Python parts wrap that up in the `machine` module.  Eg `machine/__init__.py` will do `from umachine import *` and then add everything else.  Effectively the C code is in a hidden module and the user visible module is written in pure Python.  See `uasyncio` for an example of this.

See also #5025 for an idea of a way to implement built-in functions in Python.

harbaum, 2022-02-01:
Understood, thanks for the reply. So there is currently no way to implement Counter and Encoder as documented in https://github.com/micropython/micropython/pull/8072 in Python.

I wonder if it then makes sense to put highlevel stuff like Encoder and Counter that into the machine module at all. On some plattforms without dedicated hardware counter this may actually be implemented on top of Pin.irq() which in turn can nicely be done in Python and would not require any C.

But anyway, then I'll happily support the somewhat bloated solution in https://github.com/micropython/micropython/pull/6639 . As said in the PR: Better that version than nothing.

dpgeorge, 2022-02-02:
> But what are the chances, that such a patch would be accepted upstream?

I think it's a good change.  Some ports already have `umachine` as the main name.

See #8244 for a patch that makes this change.

dpgeorge, 2022-02-02:
`umachine` is now available as of fbd47fc46cce80abfc59e680600666473a1868da

closed 2022-02-02 by dpgeorge