micropython/micropython #8242: 'newline' parameter not implemented in 'open' function.

water5, 2022-02-01:
Sometimes, communicate `.txt` file generate from between `Windows` (`\r\n`) and `MicroPython` (`\n`), will cause some problem.
[https://docs.python.org/3/library/functions.html#open-newline-parameter](https://docs.python.org/3/library/functions.html#open-newline-parameter)

dpgeorge, 2022-02-02:
It's unlikely this parameter will be implemented any time in the near future.  So in the meantime I'd suggest working around the issue by either 1) outputting `'\r\n'` explicitly from your MicroPython program (eg `print('foo', end='\r\n')`) or 2) processing the .txt files on your PC to convert them to `\r\n`.

water5, 2022-02-03:
Compare with `MicroPython` -> `CPython on Windows / POSIX (Linux)`, `CPython` could be better to handle `\n` and `\r\n`, on `MicroPython` can't recognize which files come from `Windows '\r\n'` or `POSIX(Linux) '\n'` simply.