micropython/micropython #8248: ESP32 can not work as central and peripherial at the same time

TheMrguiller, 2022-02-02:
I am working in a project and I need to have the esp32 to work as central and client at the same time. The flow of the proyect is the following:
1-My mobile phone connects to the esp32, the esp32 will be working as a peripherial.
2- Once the connection is made, I start the connection with another esp32. In this case my main esp32 is working as a central and the final one as a peripherial.
I know that the connections work fine, but every time I get to the second point the previous one disconnects. It seems it automatically disconnects, but I dont know how to solve it. I have read that is posible with arduino, but in my case I need to develop  it with python. Can someone help me?

dpgeorge, 2022-02-02:
What board are you using and what firmware version (of MicroPython)?  Can you provide example code that shows the problem?

If the hardware and IDF support being a central and peripheral at the same time, then MicroPython should be able to do it.

TheMrguiller, 2022-02-03:
@dpgeorge Right now I am using ESP32-DEVKITC-32UE and the latest micropython firmware v1.18. I have three codes for you to see:

BLE library done by me using your examples:

```
_IRQ_CENTRAL_CONNECT = const(1)
_IRQ_CENTRAL_DISCONNECT = const(2)
_IRQ_GATTS_WRITE = const(3)
_IRQ_SCAN_RESULT = const(5)
_IRQ_SCAN_DONE = const(6)
_IRQ_PERIPHERAL_CONNECT = const(7)
_IRQ_PERIPHERAL_DISCONNECT = const(8)
_IRQ_GATTC_SERVICE_RESULT = const(9)
_IRQ_GATTC_SERVICE_DONE = const(10)
_IRQ_GATTC_CHARACTERISTIC_RESULT = const(11)
_IRQ_GATTC_CHARACTERISTIC_DONE = const(12)
_IRQ_GATTC_NOTIFY = const(18)
_IRQ_GATTC_WRITE_DONE = const(17)
_FLAG_WRITE = const(0x0008)
_FLAG_NOTIFY = const(0x0010)

_ADV_IND = const(0x00)
_ADV_DIRECT_IND = const(0x01)

_UART_UUID = bluetooth.UUID("6E400001-B5A3-F393-E0A9-E50E24DCCA9E")
_UART_TX_UUID =bluetooth.UUID("6E400003-B5A3-F393-E0A9-E50E24DCCA9E")
_UART_RX_UUID =bluetooth.UUID("6E400002-B5A3-F393-E0A9-E50E24DCCA9E")
_UART_TX = (
    _UART_TX_UUID,
    _FLAG_NOTIFY,
)
_UART_RX = (
    _UART_RX_UUID,
    _FLAG_WRITE,
)
_UART_SERVICE = (
    _UART_UUID,
    (_UART_TX, _UART_RX),
)

_ADV_APPEARANCE_GENERIC_COMPUTER = const(128)#Valor para sensor de calidad es 0x0542

# Advertising payloads are repeated packets of the following form:
#   1 byte data length (N + 1)
#   1 byte type (see constants below)
#   N bytes type-specific data

_ADV_TYPE_FLAGS = const(0x01)
_ADV_TYPE_NAME = const(0x09)
_ADV_TYPE_UUID16_COMPLETE = const(0x3)
_ADV_TYPE_UUID32_COMPLETE = const(0x5)
_ADV_TYPE_UUID128_COMPLETE = const(0x7)
_ADV_TYPE_UUID16_MORE = const(0x2)
_ADV_TYPE_UUID32_MORE = const(0x4)
_ADV_TYPE_UUID128_MORE = const(0x6)
_ADV_TYPE_APPEARANCE = const(0x19)

class BLEUART:
    #Clase que inicia el objeto principal, puede actuar como cliente o server
    def __init__(self, ble,tipo ,name, rxbuf=1000):
        self._ble = ble #Copiamos el objeto BLE creado en el main
        self._ble.active(True) #Activamos BLE
        self._tipo= tipo
        self._ble.irq(self._irq) #Registramos los eventos provenientes de BLE stack
        self._handler = None
        self._rx_buffer = bytearray() #Buffer con los datos provenientes del BLE stack
        # Optionally add services=[_UART_UUID], but this is likely to make the payload too large.
        if self._tipo=="Peripheral":
            print('Peripheral')
            self._payload = advertising_payload(
[... 274 lines cut ...]
        periferico._advertise()
        periferico.irq(handler=on_rx1)
        estado="Conectado"
    if estado=="Conectado":
        pass
```

The las one is my peripherial:

```
ble= bluetooth.BLE()
name="SmartMaskPeripheral"
tipo="Peripheral"
uart= BLEUART(ble,tipo,name)
def on_rx():
    
    rx_buffer =uart.read().decode().strip() #Recibimos mensaje por blueetooth de manera asincrona
    uart.write(rx_buffer)
    print(str(rx_buffer))

uart.irq(handler=on_rx)
```

I am gonna attach for you also the output of my program. It seems that once it change from being in a central role to a peripherial role the last connection is automatically disconnected.
[image]

stephanelsmith, 2022-02-03:
Hope you are well.  I am definitely able to run BLE as both central and peripheral concurrently.  Currently have my application running in this mode.  

I am running on esp-idf v4.4rc1.  I am run on ESP32 PICO D4 chip.

TheMrguiller, 2022-02-03:
Hi @stephanelsmith,

I hope you are doing good aswell. I am new in this kind of development. Is there any kind of difference between micropython and your firmware? Do you have the code available? Does your chip have two BLE channels?

Thank you for your response

stephanelsmith, 2022-02-03:
I'm on micropython master branch and esp-idf v4.4rc1 branch.

The only thing I've done beyond normal micropython is I've modified the MP_BLUETOOTH_DEFAULT_ATTR_LEN.
https://github.com/micropython/micropython/issues/8258

I cannot share my code, but, what I've done that makes this process very simple is to use asyncio.  Notably, with asyncio I am able to remove all logic from the ble irq.  Each irq sets and async Event.  Then you can create tasks, central or peripheral, or whatever, you can control the process for each connection in a deterministic way.

TheMrguiller, 2022-02-03:
@stephanelsmith thank you so much. I didn´t know esp-idf existed and I was doing all using just micropython. I am not sure If I am able to implement it with esp-idf right now. I will try doing something similar but It is a bit overwhelming right now. If I am honest i barely know what you have done. Also I dont know anything about ESP-IDF.

dpgeorge, 2022-02-03:
The ESP-IDF is used as a low-level component of MicroPython, so you're already using it.  I think the issue is that you need a newer version of the IDF to get central and peripheral mode working together.

Can you please run `import platform` then `platform.platform()`, it will show the IDF version you use.

TheMrguiller, 2022-02-04:
@dpgeorge I am using right now this version MicroPython-1.18.0-xtensa-IDFv4.2.2-with-newlib3.0.0.  I dont know how to update the IDF version.

TheMrguiller, 2022-02-04:
I am going to close the issue. I am going to give some insight if someone like me meet with the same problem. It was a problem of misconcept:

1- I created a class that meet with the two roles, central and peripherial. These class works fine if you want it to use it separately, but if you try to use it all together with two objects it doesnt work. I dont know why but having two object one being central and another one peripherial makes everything go crazy.
2- What I have done is modify my class and use it without  differentiate the roles. If I do this everything works fine.

I wish the best for @dpgeorge and @stephanelsmith  for helping me. I am sorry for taking your precious time.

I am gonna leave the class for the public if you want to use it.
Class:
```
# Proof-of-concept of a REPL over BLE UART.
#
# Tested with the Adafruit Bluefruit app on Android.
# Set the EoL characters to \r\n.

import bluetooth
import io
import os
import micropython
import machine
from micropython import const
import struct
import time
from machine import Pin

_IRQ_CENTRAL_CONNECT = const(1)
_IRQ_CENTRAL_DISCONNECT = const(2)
_IRQ_GATTS_WRITE = const(3)
_IRQ_SCAN_RESULT = const(5)
_IRQ_SCAN_DONE = const(6)
_IRQ_PERIPHERAL_CONNECT = const(7)
_IRQ_PERIPHERAL_DISCONNECT = const(8)
_IRQ_GATTC_SERVICE_RESULT = const(9)
_IRQ_GATTC_SERVICE_DONE = const(10)
_IRQ_GATTC_CHARACTERISTIC_RESULT = const(11)
_IRQ_GATTC_CHARACTERISTIC_DONE = const(12)
_IRQ_GATTC_NOTIFY = const(18)
_IRQ_GATTC_WRITE_DONE = const(17)
_FLAG_WRITE = const(0x0008)
_FLAG_NOTIFY = const(0x0010)

_ADV_IND = const(0x00)
_ADV_DIRECT_IND = const(0x01)

_UART_UUID = bluetooth.UUID("6E400001-B5A3-F393-E0A9-E50E24DCCA9E")
_UART_TX_UUID =bluetooth.UUID("6E400003-B5A3-F393-E0A9-E50E24DCCA9E")
_UART_RX_UUID =bluetooth.UUID("6E400002-B5A3-F393-E0A9-E50E24DCCA9E")
_UART_TX = (
    _UART_TX_UUID,
    _FLAG_NOTIFY,
)
_UART_RX = (
    _UART_RX_UUID,
    _FLAG_WRITE,
)
_UART_SERVICE = (
    _UART_UUID,
    (_UART_TX, _UART_RX),
)

_ADV_APPEARANCE_GENERIC_COMPUTER = const(128)#Valor para sensor de calidad es 0x0542

# Advertising payloads are repeated packets of the following form:
#   1 byte data length (N + 1)
#   1 byte type (see constants below)
#   N bytes type-specific data

_ADV_TYPE_FLAGS = const(0x01)
_ADV_TYPE_NAME = const(0x09)
_ADV_TYPE_UUID16_COMPLETE = const(0x3)
[... 226 lines cut ...]
    _append(
        _ADV_TYPE_FLAGS,
        struct.pack("B", (0x01 if limited_disc else 0x02) +
                    (0x18 if br_edr else 0x04)),
    )

    if name:
        _append(_ADV_TYPE_NAME, name)

    if services:
        for uuid in services:
            b = bytes(uuid)
            if len(b) == 2:
                _append(_ADV_TYPE_UUID16_COMPLETE, b)
            elif len(b) == 4:
                _append(_ADV_TYPE_UUID32_COMPLETE, b)
            elif len(b) == 16:
                _append(_ADV_TYPE_UUID128_COMPLETE, b)

    # See org.bluetooth.characteristic.gap.appearance.xml
    if appearance:
        _append(_ADV_TYPE_APPEARANCE, struct.pack("<h", appearance))

    return payload
```

closed 2022-02-04 by TheMrguiller