micropython/micropython #8249: RP2: PWM duty cycle does not remain constant when frequency is changed

AlasdairAtKitronik, 2022-02-02:
The expected behaviour is that if a duty cycle value is set, this should remain constant when the frequency is varied.
In the example code (used to drive a buzzer), the duty cycle starts at 50%, but as the frrequency is increased, the duty cycle increases up to 100%. This means that the buzzer stops sounding correctly as there is no longer any PWM occurring.
```python
import machine

PWMOutput = machine.PWM(machine.Pin(0))

PWMOutput.duty_u16(32767)
freq = 500

while True:
    for freq in range (120, 5000, 50):
        PWMOutput.freq(freq)
    
    for freq in range(5000, 120, -50):
        PWMOutput.freq(freq)
```
To get round this, the cycle needs to be set ever time the frequency is set. Although this works, it does not seem the right way to be doing it.

robert-hh, 2022-02-02:
That behavior was fixed with PR #8128 early January. Updating your firmware should help.

AlasdairAtKitronik, 2022-02-02:
Brilliant, thanks for pointing me to that - code all working now.

closed 2022-02-02 by AlasdairAtKitronik