micropython/micropython #8251: esp32 esp8266 urequests

KorkhovPavel, 2022-02-02:
Hi. please tell me the urequests library, is it different for esp32 and esp8266? for esp32 it works with both http and https, and for esp8266 only http, with https error OSError 40, has anyone encountered this phenomenon? Thank you

dpgeorge, 2022-02-02:
> is it different for esp32 and esp8266?

It is the same.

> for esp8266 only http, with https error OSError 40

Can you provide a minimal example that has this error on esp8266?

KorkhovPavel, 2022-02-03:
This is the actual code I am using.
If http, everything works fine, https immediately error.
Maybe I'm doing something wrong?
Thanks for the help.

```
from machine import Pin
import dht
from urequests import post
import json

def dth11_data(url, token):
    sensor = dht.DHT11(Pin(14))
    sensor.measure()
    d = {
        "temperature": int(sensor.temperature()),
        "humidity": int(sensor.humidity())
    }
    data = json.dumps(d)

    response = post(url, data=data, headers={'Content-Type': 'application/json',
                                  'Authorization': token})
    response.close()
```

maskmly, 2022-02-04:
I have a similar problem with `urequests.get(url)` function!
If I pass a url that is not HTTPS then I can get the file. However, when I pass a url _with_ HTTPS I get the following error:
`>>> myFile = urequests.get('https://www.google.com/images/branding/googlelogo/1x/googlelogo_color_272x92dp.png')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests.py", line 116, in get
  File "urequests.py", line 62, in request
OSError: [Errno 12] ENOMEM
>>>`

Fr3nky88, 2022-02-28:
I'm following because i have the same problem with HTTPS, i fleshed the last firmware on the esp32 but still have the problem.

maskmly, 2022-04-08:
> I have a similar problem with `urequests.get(url)` function on ESP32!
> If I pass a url that is not HTTPS then I can get the file. However, when I pass a url _with_ HTTPS I get the following error:

I found a workaround for this problem. Instead of using `urequests.get(url)` I used `urequests.requests('GET', url)`.
This solved the problem with downloading from HTTPS urls.

jimmo, 2022-07-21:
This issue is now being tracked in https://github.com/micropython/micropython/issues/8940.

closed 2022-07-21 by jimmo