micropython/micropython #8254: Can a MicroPython program gets copied ?

AmirHmZz, 2022-02-02:
If I flash a program to esp32 based board, Can anyone copy my code from that board ?
I want you to answer my question in 3 scenarios :

1. Flash raw `.py` code into the board
2. Compile and flash compiled `.mpy` code to board
3. Copy scripts to micropython port for esp32 and build the whole framework with codes in it.

peterhinch, 2022-02-02:
I suggest you do a search on [the forum](http://forum.micropython.org/index.php) where this has been covered many times.

AmirHmZz, 2022-02-02:
I've searched it and found some good stuff but I still have a question,
Can a `.mpy` file gets decompiled ?

robert-hh, 2022-02-02:
That was discussed to some extend as well. As far as I recall, the answer is a strong  'maybe', and there may have been even tools mentioned. See https://forum.micropython.org/viewtopic.php?f=3&t=3868&p=55917&hilit=decompile#p55917. 
The output does not look like python (yet), but that would be a next step. As will all de-compiles, the original file cannot be restored, but something similar.

closed 2022-02-02 by dpgeorge