micropython/micropython #8256: ESP32: Can not build MicroPython with C modules.

IhorNehrutsa, 2022-02-03:
```
pc@ubuntu:~/micropython/ports/esp32$ idf.py --version
ESP-IDF v4.2.2-531-g19cca42b1f
```

```
pc@ubuntu:~/micropython/ports/esp32$ git branch
...
* master
...
...
```

```
pc@ubuntu:~/micropython/examples/usercmodule$ cd ~/micropython/ports/esp32
pc@ubuntu:~/micropython/ports/esp32$ make USER_C_MODULES=../../examples/usercmodule
idf.py -D MICROPY_BOARD=GENERIC -B build-GENERIC  -DUSER_C_MODULES=../../examples/usercmodule build
Executing action: all (aliases: build)
Running cmake in directory /home/pc/micropython/ports/esp32/build-GENERIC
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=GENERIC -DUSER_C_MODULES=../../examples/usercmodule -DCCACHE_ENABLE=0 /home/pc/micropython/ports/esp32"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /home/pc/esp-idf/components/cmock does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/pc/esp-idf/components/esp_phy does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/pc/esp-idf/components/ieee802154 does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/pc/esp-idf/components/openthread does not contain a CMakeLists.txt file. No component will be added
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/pc/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/pc/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/pc/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/pc/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/pc/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/pc/micropython/ports/esp32/build-GENERIC/sdkconfig
Loading defaults file /home/pc/micropython/ports/esp32/build-GENERIC/sdkconfig.combined...
/home/pc/micropython/ports/esp32/build-GENERIC/sdkconfig.combined:8 CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE was replaced with CONFIG_COMPILER_OPTIMIZATION_SIZE
/tmp/confgen_tmps7ka8_z7:69 line was updated to CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
/tmp/confgen_tmps7ka8_z7:70 line was updated to CONFIG_BTDM_CTRL_MODE_BTDM=n
-- Found PythonInterp: /home/pc/.espressif/python_env/idf4.2_py3.8_env/bin/python (found version "3.8.10") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- App "micropython" version: v1.18-81-g26faf74d5-dirty
-- Adding linker script /home/pc/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/pc/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/pc/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/pc/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/pc/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/pc/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/pc/micropython/ports/esp32/build-GENERIC/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/pc/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/pc/esp-idf/components/esp32/ld/esp32.peripherals.ld
Including User C Module(s) from ../../examples/usercmodule
CMake Error at /home/pc/micropython/py/usermod.cmake:42 (include):
  include could not find load file:

    ../../examples/usercmodule
Call Stack (most recent call first):
  main/CMakeLists.txt:10 (include)

Found User C Module(s): 
-- Found Python3: /home/pc/.espressif/python_env/idf4.2_py3.8_env/bin/python3.8 (found version "3.8.10") found components: Interpreter 
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/pc/esp-idf/components/app_trace /home/pc/esp-idf/components/app_update /home/pc/esp-idf/components/asio /home/pc/esp-idf/components/bootloader /home/pc/esp-idf/components/bootloader_support /home/pc/esp-idf/components/bt /home/pc/esp-idf/components/cbor /home/pc/esp-idf/components/coap /home/pc/esp-idf/components/console /home/pc/esp-idf/components/cxx /home/pc/esp-idf/components/driver /home/pc/esp-idf/components/efuse /home/pc/esp-idf/components/esp-tls /home/pc/esp-idf/components/esp32 /home/pc/esp-idf/components/esp_adc_cal /home/pc/esp-idf/components/esp_common /home/pc/esp-idf/components/esp_eth /home/pc/esp-idf/components/esp_event /home/pc/esp-idf/components/esp_gdbstub /home/pc/esp-idf/components/esp_hid /home/pc/esp-idf/components/esp_http_client /home/pc/esp-idf/components/esp_http_server /home/pc/esp-idf/components/esp_https_ota /home/pc/esp-idf/components/esp_https_server /home/pc/esp-idf/components/esp_ipc /home/pc/esp-idf/components/esp_local_ctrl /home/pc/esp-idf/components/esp_netif /home/pc/esp-idf/components/esp_ringbuf /home/pc/esp-idf/components/esp_rom /home/pc/esp-idf/components/esp_serial_slave_link /home/pc/esp-idf/components/esp_system /home/pc/esp-idf/components/esp_timer /home/pc/esp-idf/components/esp_websocket_client /home/pc/esp-idf/components/esp_wifi /home/pc/esp-idf/components/espcoredump /home/pc/esp-idf/components/esptool_py /home/pc/esp-idf/components/expat /home/pc/esp-idf/components/fatfs /home/pc/esp-idf/components/freemodbus /home/pc/esp-idf/components/freertos /home/pc/esp-idf/components/heap /home/pc/esp-idf/components/idf_test /home/pc/esp-idf/components/jsmn /home/pc/esp-idf/components/json /home/pc/esp-idf/components/libsodium /home/pc/esp-idf/components/log /home/pc/esp-idf/components/lwip /home/pc/micropython/ports/esp32/main /home/pc/esp-idf/components/mbedtls /home/pc/esp-idf/components/mdns /home/pc/esp-idf/components/mqtt /home/pc/esp-idf/components/newlib /home/pc/esp-idf/components/nghttp /home/pc/esp-idf/components/nvs_flash /home/pc/esp-idf/components/openssl /home/pc/esp-idf/components/partition_table /home/pc/esp-idf/components/perfmon /home/pc/esp-idf/components/protobuf-c /home/pc/esp-idf/components/protocomm /home/pc/esp-idf/components/pthread /home/pc/esp-idf/components/sdmmc /home/pc/esp-idf/components/soc /home/pc/esp-idf/components/spi_flash /home/pc/esp-idf/components/spiffs /home/pc/esp-idf/components/tcp_transport /home/pc/esp-idf/components/tcpip_adapter /home/pc/esp-idf/components/tinyusb /home/pc/esp-idf/components/ulp /home/pc/esp-idf/components/unity /home/pc/esp-idf/components/vfs /home/pc/esp-idf/components/wear_levelling /home/pc/esp-idf/components/wifi_provisioning /home/pc/esp-idf/components/wpa_supplicant /home/pc/esp-idf/components/xtensa
-- Configuring incomplete, errors occurred!
See also "/home/pc/micropython/ports/esp32/build-GENERIC/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
make: *** [Makefile:34: all] Error 2

```

IhorNehrutsa, 2022-02-03:
wrong
```
pc@ubuntu:~/micropython/ports/esp32$ make USER_C_MODULES=../../examples/usercmodule/micropython.cmake
```

right
```
pc@ubuntu:~/micropython/ports/esp32$ make USER_C_MODULES=../../../examples/usercmodule/micropython.cmake
```

dpgeorge, 2022-02-03:
Different ports need different paths, some need `../..` and some need `../../..`.

For the docs it might be better to use absolute paths instead of relative.  That should always work.

closed 2025-02-04 by IhorNehrutsa