micropython/micropython #8258: mtu vs MP_BLUETOOTH_DEFAULT_ATTR_LEN

stephanelsmith, 2022-02-03:
Hope all is well.  Reporting a snag for me which may or may not be an actual issue.  Either way, I have a work-around, but wanted to raise.

After configuring MTU, **ble.gatts_notify** is using the new mtu, writing mtu-3 bytes. 
However, **ble.gattc_write** still writes 20 bytes max.

Process (brief):
1) activate ble
2) setting MTU  `ble.config(mtu = xxx)`
3) setting rxbuf  `ble.config(rxbuf = xxx)`
4) exchanging MTU  `ble.gattc_exchange_mtu(conn_handle)`
5) confirmed w/ **IRQ_MTU_EXCHANGED**

The work-around for me is simple.  In **mpconfigboard.h**, I've added:
`#define MP_BLUETOOTH_DEFAULT_ATTR_LEN (xxx)`

As long as **MP_BLUETOOTH_DEFAULT_ATTR_LEN** is greater than the mtu, the payload is the write size for both **ble.gatts_notify** and **ble.gattc_write**.  Side-effects to be concerned about?

I took a poke around to see if I could find anything obvious to my eyes.  I played around with exchanging the MTU before discovering characteristics in case there's a race condition or order of operation.  No effect.

stephanelsmith, 2022-02-03:
Related
https://github.com/micropython/micropython/issues/6947

MP_BLUETOOTH_DEFAULT_ATTR_LEN  defined here:
https://github.com/micropython/micropython/blob/master/extmod/modbluetooth.h

dpgeorge, 2022-02-03:
Did you try `BLE.gatts_set_buffer()`?  http://docs.micropython.org/en/latest/library/bluetooth.html#bluetooth.BLE.gatts_set_buffer

stephanelsmith, 2022-02-04:
PERFECT.  I missed that and sincerely appreciate you brining it to my attention.  That's the ticket!

On MTU exchange, I can upsize the specific handle buffers to the appropriate size, in my case, matching the MTU so gattc_write and gatts_notify operate symmetrically.

closed 2022-02-04 by stephanelsmith