micropython/micropython #8260: ESP32: recent IDF versions build woes (crashing, missing memory, iram0 overflow…)

jonathanhogg, 2022-02-04:
I'm seeing a bunch of problems building against different versions of the IDF. All of the following is building 102cc12dbbe10c3caf3e1e141be4ab4366dbd02f with 2 of my patches integrated on top (#6263 and #7582).

A `GENERIC` ESP32 build against the latest v4.4 tag results in a binary that explodes on boot:

```
ELF file SHA256: 2995987005bacb8b

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4344
load:0x40078000,len:13816
load:0x40080400,len:3340
0x40080400: _init at ??:?

entry 0x40080618
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4014899a  PS      : 0x00060733  A0      : 0x800d7048  A1      : 0x3ffcef20  
0x4014899a: uart_isr_register at /Users/jonathan/Projects/outputarts/esp-idf-4.4/components/driver/uart.c:594

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb78e8  A5      : 0x0000040e  
A6      : 0x3ffcef50  A7      : 0x3ffbf24c  A8      : 0x00000000  A9      : 0x3ffceef0  
A10     : 0x00000000  A11     : 0x0000040e  A12     : 0x4008334c  A13     : 0x00000000  
0x4008334c: uart_irq_handler at /Users/jonathan/Projects/outputarts/micropython/ports/esp32/uart.c:46

A14     : 0x3ffcef50  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000008  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x40148997:0x3ffcef200x400d7045:0x3ffcef50 0x400d6eb9:0x3ffcef80 
0x40148997: uart_isr_register at /Users/jonathan/Projects/outputarts/esp-idf-4.4/components/driver/uart.c:594

0x400d7045: uart_init at /Users/jonathan/Projects/outputarts/micropython/ports/esp32/uart.c:41

0x400d6eb9: mp_task at /Users/jonathan/Projects/outputarts/micropython/ports/esp32/main.c:96
```

Switching to v4.3.2 works, but I am unable to do a matching `GENERIC_SPIRAM` build (for the other boards that I use) as it seems to be overflowing the `iram0` memory segment:

```
[100%] Linking CXX executable micropython.elf
/Users/jonathan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: micropython.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/Users/jonathan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/Users/jonathan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 492 bytes
collect2: error: ld returned 1 exit status
make[3]: *** [micropython.elf] Error 1
make[2]: *** [CMakeFiles/micropython.elf.dir/all] Error 2
make[1]: *** [all] Error 2
make failed with exit code 2
make: *** [all] Error 2
```

I have verified that this is the same building against `master` without my patches.

This last error is a little outside my area of expertise: I'm not actually familiar with the ESP32/MicroPython memory model. I did a little hunting to see if I could figure out what's in that `text` area and why it's suddenly overflowing with v4.3.2, but nothing immediately jumped out at me. I'm also not sure how the size of the regions is configured.

Backing up one more version to build against v4.3.1 appears to work OK, but suffers from the missing memory problem documented in #7813 

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:13396
load:0x40080400,len:3312
entry 0x4008061c
MicroPython v1.18-82-gd57e23b8d on 2022-02-04; ESP32 module with ESP32
Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
63008
```

This is about 109000 free on a straight MicroPython 1.18 GENERIC download.

Going back to v4.0.2 I get seemingly good builds for both `GENERIC` and `GENERIC_SPIRAM` and my memory back:

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4252
load:0x40078000,len:11920
load:0x40080400,len:3344
entry 0x4008060c
MicroPython v1.18-82-gd57e23b8d on 2022-02-04; ESP32 module with ESP32
Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
110176
```

Is the moral of this tale: *do not attempt to build MicroPython against any version of the IDF more recent than v4.0.2*?

I've been using v4.3.1 for ages, but always with a `GENERIC_SPIRAM` build so I hadn't noticed the missing memory on a non-SPIRAM device until yesterday.

stephanelsmith, 2022-02-04:
I'm successfully building esp-idf v4.4rc1 with micropython master.

russhughes, 2022-02-05:
I don't know if this us useful but here is an idf.py size-components diff report between GENERIC_SPIRAM and GENERIC using esp-idf V4.4 showing the static IRAM overflow.

```
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in idf_size.py

<CURRENT> MAP file: build-GENERIC_SPIRAM/micropython.map
<REFERENCE> MAP file: build-GENERIC/micropython.map

Difference is counted as <CURRENT> - <REFERENCE>,           i.e. a positive number means that <CURRENT> is larger.
Total sizes of <CURRENT>:                                       <REFERENCE>     Difference
Used static DRAM:   51492 bytes (  73088 remain, 41.3% used)          51320           +172 (   -172 remain,      +0 total)
      .data size:   22564 bytes                                       22448           +116
      .bss  size:   28928 bytes                                       28872            +56
Used static IRAM:  132830 bytes (  -1758 remain, 101.3% used) Overflow detected!         118855         +13975 ( -13975 remain,      +0 total)
      .text size:  131803 bytes                                      117828         +13975
   .vectors size:    1027 bytes                                        1027
Used stat D/IRAM:  184322 bytes (  71330 remain, 72.1% used)         170175         +14147 ( -14147 remain,      +0 total)
      .data size:   22564 bytes                                       22448           +116
      .bss  size:   28928 bytes                                       28872            +56
      .text size:  131803 bytes                                      117828         +13975
   .vectors size:    1027 bytes                                        1027
Used Flash size : 1556802 bytes                                     1547302          +9500
      .text     : 1139527 bytes                                     1087411         +52116
      .rodata   :  417019 bytes                                      459635         -42616
Total image size: 1867590 bytes (.bin may be padded larger)         1829908         +37682
Per-archive contributions to ELF file:
            Archive File              DRAM .data               .rtc.data               DRAM .bss               .rtc.text             IRAM0 .text             & 0.vectors            ram_st_total             Flash .text               & .rodata        & .rodata_noload              & .appdesc             flash_total
                        |    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>|    <C>|    <R>|<C>-<R>
                         -----------------------                         -----------------------                         -----------------------                         -----------------------                         -----------------------                         -----------------------                        
               libmain.a|    146|    146|       |   2054|   2054|       |   2406|   2406|       |       |       |       |   8524|   8177|   +347|       |       |       |  13130|  12783|   +347| 239554| 228858| +10696| 255044| 303226| -48182|       |       |       |       |       |       | 505322| 542461| -37139
               liblwip.a|     21|     21|       |       |       |       |   3888|   3888|       |       |       |       |       |       |       |       |       |       |   3909|   3909|       | 120345| 111234|  +9111|  17728|  17728|       |       |       |       |       |       |       | 138094| 128983|  +9111
           libnet80211.a|    759|    759|       |       |       |       |   6085|   6085|       |       |       |       |       |       |       |       |       |       |   6844|   6844|       | 103797| 103752|    +45|   6014|   6014|       |  12516|  12516|       |       |       |       | 123086| 123041|    +45
           libbtdm_app.a|    773|    773|       |       |       |       |   2617|   2617|       |       |       |       |  28344|  28416|    -72|       |       |       |  31734|  31806|    -72|  77146|  77138|     +8|   5207|   5207|       |       |       |       |       |       |       | 111470| 111534|    -64
             libdriver.a|    404|    404|       |       |       |       |    408|    408|       |       |       |       |   7159|   6882|   +277|       |       |       |   7971|   7694|   +277|  61940|  58380|  +3560|  21958|  21958|       |       |       |       |       |       |       |  91461|  87624|  +3837
         libmbedcrypto.a|     64|     64|       |       |       |       |     85|     85|       |       |       |       |     30|     30|       |       |       |       |    179|    179|       |  73675|  70420|  +3255|  10869|  10869|       |       |       |       |       |       |       |  84638|  81383|  +3255
                 libbt.a|   4923|   4923|       |       |       |       |   5226|   5226|       |       |       |       |   1782|   1702|    +80|       |       |       |  11931|  11851|    +80|  64151|  59031|  +5120|   5802|   5802|       |       |       |       |       |       |       |  76658|  71458|  +5200
                  libc.a|     41|      4|    +37|       |       |       |       |       |       |       |       |       |   6648|       |  +6648|       |       |       |   6689|      4|  +6685|  60546|  55745|  +4801|   4206|   4206|       [... cut ...]
                libcxx.a|       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |     41|     41|       |       |       |       |       |       |       |       |       |       |     41|     41|       
               libmesh.a|       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       |       

```

jonathanhogg, 2022-02-05:
Interesting! Thanks for this @russhughes.

Inspired by this I've used `idf_size.py` to do a comparison of 4.3.1 and 4.3.2 `GENERIC_SPIRAM` builds (the latter failing to link) and these are the main changes in `.iram0.text`:

```
libbootloader_support.a:bootloader_flash.c.o        710
libnewlib.a:assert.c.o                              362
libesp_hw_support.a:regi2c_ctrl.c.o                 171
libesp_system.a:sleep_modes.c.o                     163
libbtdm_app.a:coex_schm.o                           160
libbtdm_app.a:arch_main.o                           148
libmain.a:vm.c.o                                    123
libhal.a:spi_hal_iram.c.o                            80
libfreertos.a:tasks.c.o                              62
libdriver.a:gpio.c.o                                 40
...
libbtdm_app.a:lld_pdu.o                              -4
libbtdm_app.a:ld_acl.o                               -4
libbtdm_app.a:huart.o                                -4
libc.a:lib_a-strcmp.o                                -8
libesp_timer.a:esp_timer_impl_lac.c.o               -15
libesp_timer.a:esp_timer.c.o                        -15
libesp_system.a:cpu_start.c.o                       -28
libbtdm_app.a:lld_evt.o                             -52
libesp_hw_support.a:rtc_clk.c.o                     -76
libbtdm_app.a:ld_fm.o                              -244
```

The worst culprit, `bootloader_flash.c.o`, appears to have gained a load of new functions with `IRAM_ATTR`, which I assume is pulling the code for those into SRAM.

The `GENERIC_SPIRAM` build appears to just generally use more SRAM than `GENERIC` (I guess it can afford to be profligate) and that combined with these changes is busting the segment.

dpgeorge, 2022-02-06:
There are at least 2 factors here for running out of iRAM:
- Building with SPIRAM enabled includes some additional compiler flags to work around bugs in the silicon/caching.  These options add extra memory barrier instructions in many places, and this increases the size of the firmware, including functions that live in iRAM.  Overall the firmware is just bigger with SPIRAM enabled (maybe this can be improved with variants like C3?  not sure).
- In commit 549448e8bbc8ce0b6b5fc51c0660acdaff18c3d6 some functions in the MP runtime were placed in iRAM to improve performance (also -O2 was used which increase build size cf -Os).  You can try switching back to -Os (`GENERIC_D2WD` board does this to reduce its size), and/or disabling those `MICROPY_WRAP_xxx` macros so that iRAM has less in it.

robert-hh, 2022-02-06:
My next step was trying the MIMXRT1050 board, and on this board the SAI_RX_SYNC signal is NOT wired to the Codec. So I had to go back using the TX_SYNC signal for recording. Since that works on the 1020 as well, we should better use that setting as default with the WM8960 codec. That will work on all boards. The Codec driver has an option to switch that, so it's not hardwired.

jonathanhogg, 2022-02-06:
Thanks for the background @dpgeorge. I just attempted a new 4.3.2 `GENERIC_SPIRAM_SMALL` build – changing just the optimization to be for size rather than performance – and that results in a successful link.

For interest, changes in `.iram0.text` usage between this and the standard (failing) 4.3.2 `GENERIC_SPIRAM` build are:

```
## -- End pasted text --
libmain.a:scheduler.c.o                              17
libesp32.a:spiram.c.o                                12
libesp_timer.a:esp_timer_impl_lac.c.o                10
libheap.a:multi_heap.c.o                              8
libesp_timer.a:ets_timer_legacy.c.o                   6
libesp_hw_support.a:cpu_util.c.o                      4
libspi_flash.a:memspi_host_driver.c.o                 3
libnewlib.a:locks.c.o                                 3
libhal.a:soc_hal.c.o                                  3
libesp_system.a:system_api.c.o                        2
...
libmain.a:vm.c.o                                   -231
libhal.a:spi_flash_hal_iram.c.o                    -242
libesp_hw_support.a:rtc_clk.c.o                    -257
libesp_system.a:panic.c.o                          -259
libesp_system.a:sleep_modes.c.o                    -268
libdriver.a:rmt.c.o                                -292
libhal.a:spi_hal_iram.c.o                          -341
libspi_flash.a:spi_flash_chip_generic.c.o          -354
libesp32.a:spiram_psram.c.o                        -535
libfreertos.a:tasks.c.o                            -622
```

So quite a surprising number of large reductions. Off to dig out a microcontroller to test this build on…

UnexpectedMaker, 2022-02-08:
I'm also seeing boot loop issues with latest master and IDF4.4 release on my TinyPICO (has SPIRAM) - working fine on my S2 and S3 boards.

So it seems we will need to remove some of the IRAM optimisations for PSRAM boards @dpgeorge ? Is that even possible to do on a per board level?

dpgeorge, 2022-02-08:
@UnexpectedMaker can you test the plain `GENERIC` build with IDF4.4 on TinyPICO?  Then we can narrow down SPIRAM vs non-SPIRAM builds.

> So it seems we will need to remove some of the IRAM optimisations for PSRAM boards @dpgeorge ? Is that even possible to do on a per board level?

Yes.  Define relevant macros in `mpconfigboard.h`, eg `#define MICROPY_WRAP_MP_BINARY_OP (0)`

UnexpectedMaker, 2022-02-08:
GENERIC still boot loops my TinyPICO.

jonathanhogg, 2022-02-08:
Yes, at the moment I'm not convinced that the boot loop failure on v4.4 tag builds is the same IRAM problem as the other stuff I've been hitting. The backtrace I got suggested a bad pointer in the UART initialisation. I've not had time to dig deeper into it though – I've gone back to 4.3.2 and am building `GENERIC` as normal and a new `-Os` version of `GENERIC_SPIRAM`.

jonathanhogg, 2022-02-08:
[Focusing on the main issue from all of this noise that I've generated:]

Given there's a clear constraint on `iram0` size with later versions of the IDF, is there a way of having the build variables switch on the version and build with `-Os`? Or is it possible to build the IDF components with `-Os` and build MicroPython with `-O2`? I am insufficiently familiar with the CMake build process to play with this.

jprodriguez-nbs, 2022-02-25:
About the UART problem signaled by jonathanhogg, I narrowed it down to the commit 219242de in esp-idf
That commit has a change to be able to register a new ISR and unregister a previous ISR.
If you checkout commit d83021a6, you can avoid the UART problem.

Besides this, network.PPP was also working on old versions (T-CALL with SIM800L, micropython 1.17, esp-idf 4.3.2) and no longer works with micropython 1.18, esp-idf v4.4.

dpgeorge, 2022-04-14:
GENERIC_SPIRAM and UM_TINYPICO (which also uses SPIRAM) are now unconditionally built with -Os instead of -O2, as of ae3d5289a5fa893ed9fd9c6d824043d7644179d7.  I think that's a reasonable thing to do, otherwise things get too complicated.

This issue should be able to be closed now.

closed 2022-04-14 by dpgeorge

harbaum, 2023-12-05:
I am seeting the same again at this very moment. A clean micropython pull and esp-idf 5.0.2 (which is imho the version currently supposed to use) results in:

`region `iram0_0_seg' overflowed by 432 bytes`

robert-hh, 2023-12-05:
> region iram0_0_seg' overflowed by 432 bytes`

I do not get that with a GENERIC_ESP32 build. Which board variant did you try?

harbaum, 2023-12-05:
Ah, sorry. It's with make
`BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM`

Going back to tag v1.21.0 fixes this. But current git fails.

robert-hh, 2023-12-05:
Confirmed. The SPIRAM board variant fails to build. It came with commit:
`30b0ee34d esp32: Enable mbedtls cert time validation.`

dpgeorge, 2023-12-05:
See #13131.