micropython/micropython #8266: Why does micropython/ports/esp32/partitions-16MiB.csv stops at 15 MB?

gmos, 2022-02-06:
Why does micropython/ports/esp32/partitions-16MiB.csv stops at 15 MB? 

The filename suggest that 16 MiB is used but in fact it stops at 15.   The 16MiB-ota versioin does use the full 16MB,

dpgeorge, 2022-02-07:
It misses the last 680064 bytes, so uses about 15.3MiB.

I honestly don't know why I used `0xD59F80` as the size of the `vfs` partition.... this is equal to 14000000 bytes so I guess I was thinking that there is 2MiB for firmware and 14MiB for filesystem, and then put in 14,000,000 instead of 14 * 1024 * 1024.

So that 0xD59F80 value just needs to be changed to 0xE00000.

gmos, 2022-02-07:
Right Damien.  Do you want a pull request?

dpgeorge, 2022-02-08:
Thanks, but I fixed it in 6fe3856c4f5baeb5ef7b46dbdfd99bf34f967169

Thanks for the report!

closed 2022-02-08 by dpgeorge