micropython/micropython #8267: [ESP32] ADC atten does not set on init, does set when calling method

cmoevus, 2022-02-06:
Hi,

I have noticed that, for example, the following code 
```
p = ADC(Pin(35), atten=ADC.ATTN_11DB)
p.read()
```
will not actually set the attenuation, whereas 
```
p = ADC(Pin(35))
p.atten = ADC.ATTN_11DB
p.read()
```
does.

This is using a ESP-WROOM-32 with micropython 1.18.

Thanks!

dpgeorge, 2022-02-07:
Can you please try with the latest master (unstable build at `micropython.org/download`)?  This issue is probably fixed by 63438a31bb6d0c8012e338433df95700f1cbca97

dpgeorge, 2022-02-08:
This has now been fully tested on the latest master, and a bug fixed with initialisation of `atten` in 468d1979baf022b9598137243007cc550ff0b306

I'll close this issue.  Pleas reopen if the issue persists.

closed 2022-02-08 by dpgeorge