micropython/micropython #8272: RFC: implement rsync on mpremote?

peterhinch, 2022-02-07:
Given the recent improvements to **mpremote** I thought I'd throw in some user feedback.

I have a project comprising ~112 Python files. These are organised in Python packages with nested subdirectories. Formerly I developed using a modified version of **rshell**, using `rsync` and per-project text macros to ensure that the target was always running current code.

With **mpremote** I mount the project root directory on the PC so there is no issue of synchronisation. Now that paste mode works I can use **mpremote** for almost everything. As a development tool it is superb :)

Alas **rshell** remains the quickest way to develop on platforms where the USB link is slow. Synchronising files on the target and running the application can be substantially quicker than running a large application from a mounted PC directory. This is because the amount of USB transfer is typically much reduced.

An `rsync` command, combined with simple per-project Bash scripts, would provide all the functionality of **rshell** with macros.

Note that **rshell** uses [timestamps](https://github.com/dhylands/rshell/blob/b0acaafa74878269458688965ec29e98ceaf1d16/rshell/main.py#L1002) to do synchronisation: it ensures that this works by synchronising the target's clock to the PC's when **rshell** connects.

leycec, 2022-02-22:
Tangentially, [the official documentation for `mpremote mount` is crazily terse](https://docs.micropython.org/en/latest/reference/mpremote.html):

> mount the local directory on the remote device:

**...that's it.** That's the entire documentation for (arguably) the most useful `mpremote` subcommand. 

In our case, we had no idea what "mount" even meant in the context of a bare-metal port lacking an actual filesystem – so we ignored `mount` entirely in favour of the dramatically slower `cp` subcommand until finally tripping across this issue and [this relevant six-month-old forum thread](https://forums.raspberrypi.com/viewtopic.php?t=314259). The feeling of exhaustion mingled with relief is real. :face_exhaling: 

Since `mount`-based execution is now within the same order of magnitude as Flash-based execution, this kinda feels more like a *documentation* issue than anything. While an `rshell`-style `rsync` subcommand would also be useful, everyone would probably get the most mileage out of official documentation that meaningfully describes `mount`, promotes its efficiency, and encourages its use everywhere.

Forum user fivdi [concisely nailed it with this description](https://forums.raspberrypi.com/viewtopic.php?p=1880987&sid=6809b165e1d1cff6f259bf3e570eebe8#p1880987):

> The command that I like most so far is `mpremote mount .` which makes it look as if the current directory on the development machine had been copied to the microcontroller flash, but it hasn't, and everything works as if it had been copied.

That's actually pretty mind-blowing and deserves to be better highlighted. :exploding_head:

leycec, 2022-02-22:
**Relatedly:** is there a corresponding `mpremote umount` subcommand or equivalent idiom for achieving the same effect of resetting the visible filesystem back to Flash?

There's *got* to be a more scripting-friendly approach than hard or soft resets.

robert-hh, 2022-02-22:
> s there a corresponding mpremote umount subcommand or equivalent idiom for achieving the same effect of resetting the visible filesystem back to Flash?

The flash file system is available under the path /flash, the remote file system under /remote. So you can access the files in flash or use os.chdir() to change the path.

dpgeorge, 2022-02-22:
I do think `mpremote rsync` would be a good addition.  It would fit it well with the other filesystem commands (ls, cp, mkdir, etc) and definitely be useful for deployment.

PR #8231 looks like it includes some functions which could be used for rsync'ing.

I'd also like to improve the overall speed of file transfers, and remote mounting, if possible.

andrewleech, 2023-04-03:
FWIW my manifest integration into mpremote included a sync style of working: https://gitlab.com/gitlab-org/gitlab/-/issues/404326

peterhinch, 2023-04-03:
This sounds great so long as it's an option. I use two styles of working, depending on the project.
 1. `mpremote` Mounting the PC filesystem. A major benefit is that it leaves the hardware filesystem untouched.
 2. `rshell` Using rsync to deploy. Rather slow. A fast `mpremote` version would be excellent.

I think it's important to support both.