micropython/micropython #8275: ADC reading is returning broken or incorrect values (All ESP32 variants)

UnexpectedMaker, 2022-02-08:
In current master, all ADC read and read_uv results are returning incorrect values on all ESP32 variants.... or just broken, like on the S3.

On ESP32, reading a 1.1V value on an ADC pin now returns 3510-3545 range - instead of 4096.
In 1.18 release version, reading 1.1V on an ADC pin returned a solid 4096, as it should for max value.

On ESP32-S2, reading a 1.1V value on an ADC now returns MASSIVELY swinging values...   all over the place. Yes, the S2 ADC is crapy and unstable, even in Arduino, but 1.18 release has a dramatically less swing to the returned values and they much closer to the 8192 they should be.

On ESP32-S3, reading the ADC only returns a value the first read, then returns 0 from them on. The value returned from 1.1V is totally wrong... I get 455 or 470 - instead of 4096.

read_uv() also suffers from the same issue on the S3, where you 1 read only, then 0 for any other read.

Soft reset doesn't fix this. Hard reset does, pits you back to *some value* on first read, then 0.

This was not the case on S3 pre machine_adcblock being merged. 

This is all built against IDF 4.4 release version:
ESP-IDF v4.4-98-geb3797dc3f

UnexpectedMaker, 2022-02-08:
Ok, I just completely stomped my MP and IDF repos and re-cloned and built against them.

ESP32-S3 now doesn't do the first read only issue - it reads every time now (no idea why) but the value is not correct.

In `machine_adblock.c` DEFAULT_VREF is set to 1100 - and I am reading 1.1V at my ADC pin on my DMM, but adc.read() returns 1109 and adc.read_uv() returns 27000 ish.

```
MicroPython v1.18-100-ge88430d9d-dirty on 2022-02-08; TinyS3 with ESP32-S3-FN8
Type "help()" for more information.

>>> from machine import Pin, ADC
>>> adc = ADC(Pin(10))
>>> adc.read()
1166
>>> adc.read()
1134
>>> adc.read()
1162
>>> adc.read()
1164
>>> adc.read_uv()
264000
>>> adc.read_uv()
270000
>>> adc.read_uv()
270000
>>> adc.read()
1127
>>> adc.read()
1109
>>> adc.read()
1107
>>> adc.read_uv()
262000
>>> adc.read_uv()
255000
>>> adc.read_uv()
263000
>>> adc.read_uv()
258000
```
adc.read() should be returning 4096

Still have the same issues with ESP32 (TinyPICO) reading 3500 range.

```
MicroPython v1.18-100-ge88430d9d-dirty on 2022-02-08; TinyPICO with ESP32-PICO-D4
Type "help()" for more information.
>>> from machine import Pin, ADC
>>> adc = ADC(Pin(35))
>>> adc.read()
3521
>>> adc.read()
3519
>>> adc.read()
3519
>>> adc.read()
3519
```

dpgeorge, 2022-02-08:
For esp32, it looks like the ADC attenuation is not being set if it is the default.

UnexpectedMaker, 2022-02-08:
It looks like it's being set in `madc_init_helper` - which is called in `madc_make_new`

dpgeorge, 2022-02-08:
See #8276 for a fix to setting the atten value.

jonathanhogg, 2022-02-08:
Ah. Sorry again. This is my mistake translating from my original code (all ADC objects in RAM) to the optimised version (only attenuations stored in RAM) – the `-1` sentinel made sense when I could set it in the initialiser.

Oddly I haven't noticed this bug in fairly extensive actual usage on my devices – I wonder if this is just because I never use a soft reset…

dpgeorge, 2022-02-08:
> Oddly I haven't noticed this bug in fairly extensive actual usage on my devices – I wonder if this is just because I never use a soft reset…

It's not soft reset, the issue comes about when the ADC object is constructed without specifying `atten`.  So I guess you always specify `atten=` in your use of it.

dpgeorge, 2022-02-08:
Fixed by 468d1979baf022b9598137243007cc550ff0b306

closed 2022-02-08 by dpgeorge