micropython/micropython #8277: ESP32: instant crash of `GENERIC` build against IDF v4.4

jonathanhogg, 2022-02-08:
A `GENERIC` build against the IDF v4.4 (release tag) is instantly exploding registering the UART ISR:

```
entry 0x40080618
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4014899a  PS      : 0x00060733  A0      : 0x800d7048  A1      : 0x3ffcef20  
0x4014899a: uart_isr_register at /Users/jonathan/Projects/outputarts/esp-idf-4.4/components/driver/uart.c:594

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb78c8  A5      : 0x0000040e  
A6      : 0x3ffcef50  A7      : 0x3ffbf24c  A8      : 0x00000000  A9      : 0x3ffceef0  
A10     : 0x00000000  A11     : 0x0000040e  A12     : 0x4008334c  A13     : 0x00000000  
0x4008334c: uart_irq_handler at /Users/jonathan/Projects/outputarts/micropython/ports/esp32/uart.c:46

A14     : 0x3ffcef50  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000008  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x40148997:0x3ffcef200x400d7045:0x3ffcef50 0x400d6eb9:0x3ffcef80 
0x40148997: uart_isr_register at /Users/jonathan/Projects/outputarts/esp-idf-4.4/components/driver/uart.c:594

0x400d7045: uart_init at /Users/jonathan/Projects/outputarts/micropython/ports/esp32/uart.c:41

0x400d6eb9: mp_task at /Users/jonathan/Projects/outputarts/micropython/ports/esp32/main.c:96
```

Following that traceback, that's the `uart_isr_register()` call here:

```C
void uart_init(void) {
    uart_isr_handle_t handle;
    uart_isr_register(UART_NUM_0, uart_irq_handler, NULL, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM, &handle);
    uart_enable_rx_intr(UART_NUM_0);
}
```

and the `...->intr_handle = *handle;` line here:

```C
esp_err_t uart_isr_register(uart_port_t uart_num, void (*fn)(void *), void *arg, int intr_alloc_flags,  uart_isr_handle_
t *handle)
{
    int ret;
    ESP_RETURN_ON_FALSE((uart_num < UART_NUM_MAX), ESP_FAIL, UART_TAG, "uart_num error");
    UART_ENTER_CRITICAL(&(uart_context[uart_num].spinlock));
    ret = esp_intr_alloc(uart_periph_signal[uart_num].irq, intr_alloc_flags, fn, arg, handle);
    if (ret == ESP_OK) {
        p_uart_obj[uart_num]->intr_handle = *handle;
    }
    UART_EXIT_CRITICAL(&(uart_context[uart_num].spinlock));
    return ret;
}
```

I guess that suggests that the `p_uart_obj[uart_num]` pointer is invalid.

The only place I can see this being set is in the `uart_driver_install()` function. This is being called in `machine_uart.c` and in `btstack_stdin_esp32.c`, but I see no sign of it being called in `uart.c`. The IDF UART documentation seems to have always said that this should be called prior to using the UART, so it's not a new v4.4 thing. Presumably something else was calling it behind the scenes and that is now no longer happening.

dpgeorge, 2022-02-08:
This is related to #8255.

jonathanhogg, 2022-02-08:
It seems that this storing of the handle in the `p_uart_obj` is new to v4.4:

```diff
@@ -594,9 +587,12 @@ esp_err_t uart_enable_tx_intr(uart_port_t uart_num, int enable, int thresh)
 esp_err_t uart_isr_register(uart_port_t uart_num, void (*fn)(void *), void *arg, int intr_alloc_flags,  uart_isr_handle_t *handle)
 {
     int ret;
-    UART_CHECK((uart_num < UART_NUM_MAX), "uart_num error", ESP_FAIL);
+    ESP_RETURN_ON_FALSE((uart_num < UART_NUM_MAX), ESP_FAIL, UART_TAG, "uart_num error");
     UART_ENTER_CRITICAL(&(uart_context[uart_num].spinlock));
     ret = esp_intr_alloc(uart_periph_signal[uart_num].irq, intr_alloc_flags, fn, arg, handle);
+    if (ret == ESP_OK) {
+        p_uart_obj[uart_num]->intr_handle = *handle;
+    }
     UART_EXIT_CRITICAL(&(uart_context[uart_num].spinlock));
     return ret;
 }
```

I tried a quick hack of adding:

```C
    uart_driver_install(UART_NUM_0, 256, 256, 0, NULL, 0);
```

into the `uart_init()` function, and it fixes the instant crash, but I'm not getting any serial input so I suspect this needs more initialisation to be done by someone who's read the docs more fully than I have.

Gonna leave this for @andrewleech to think about ;-)

andrewleech, 2022-02-24:
I expect that the current commit / version of https://github.com/micropython/micropython/pull/8279 will resolve this debf9d13837473de97eba06e0912f55fe4179308

dpgeorge, 2022-03-01:
This should be fixed by 2cc9232781d818b8a317dc7b173ecd5b752867e7 (I tested this with the GENERIC build on a TinyPICO and it works).

closed 2022-03-01 by dpgeorge

jonathanhogg, 2022-03-01:
Can confirm that a v4.4 build works fine on my devices too.