micropython/micropython #8283: ESP32: `Pin.PULL_HOLD` does not do what the quickref says it does

jonathanhogg, 2022-02-09:
The ESP32 quickref says this about `PULL_HOLD` in [Pins and GPIO](http://docs.micropython.org/en/latest/esp32/quickref.html#pins-and-gpio):

> - The pull value of some pins can be set to `Pin.PULL_HOLD` to reduce power consumption during deepsleep.

and this in [Deep-sleep Mode](http://docs.micropython.org/en/latest/esp32/quickref.html#deep-sleep-mode):

> - There may be some leakage current flowing through enabled internal pullups. To further reduce power consumption it is possible to disable the internal pullups:
> 
> ```
> p1 = Pin(4, Pin.IN, Pin.PULL_HOLD)
> ```

Looking at the code that implements this I see:

```C
        if (mode & GPIO_PULL_HOLD) {
            gpio_hold_en(self->id);
        } else if (GPIO_IS_VALID_OUTPUT_GPIO(self->id)) {
            gpio_hold_dis(self->id);
        }
```

but the [IDF documentation for `gpio_hold_en`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#_CPPv412gpio_hold_en10gpio_num_t) says:

> Enable gpio pad hold function.
> 
> The gpio pad hold function works in both input and output modes, but must be output-capable gpios. If pad hold enabled: in output mode: the output level of the pad will be force locked and can not be changed. in input mode: the input value read will not change, regardless the changes of input signal.
> 
> The state of digital gpio cannot be held during Deep-sleep, and it will resume the hold function when the chip wakes up from Deep-sleep.

I've verified this locking behaviour in an actual device for outputs using `pull=Pin.PULL_HOLD`: subsequent calls to `.value(0|1)` do nothing. Curiously, I do **not** see this locking behaviour for input-configured pins: I can change the applied voltage and `.value()` will return either 0 or 1 – I need to do some more digging on this.

I've also confirmed that the output drivers and input pull-ups/-downs disconnect when going into deep-sleep, leaving the GPIO floating whether the pin is held or not. On wake, a held pin will still be locked but the drive and pull status is **not** restored so the pin remains floating. I don't really understand the purpose of this floating-but-held-on-wake behaviour.

However, the IDF docs go on to say:

>  If the digital gpio also needs to be held during Deep-sleep, `gpio_deep_sleep_hold_en` should also be called.

which is the behaviour that I'm *actually* interested in. So I added a new `gpio_deep_sleep_hold(True|False)` function to the `esp32` module that calls this, and then confirmed that:

```python
from machine import Pin, deepsleep
from esp32 import gpio_deep_sleep_hold

Pin(22, Pin.OUT, value=1, pull=Pin.PULL_HOLD)
gpio_deep_sleep_hold(True)
deepsleep(5000)
```

will maintain the drive value on the pin when the chip enters deep-sleep and on wake. This also works for pull-ups and -downs, a la:

```python
from machine import Pin, deepsleep
from esp32 import gpio_deep_sleep_hold

Pin(22, Pin.IN, pull=Pin.PULL_HOLD|Pin.PULL_UP)
gpio_deep_sleep_hold(True)
deepsleep(5000)
```

So, in short:

- `gpio_hold_en()` only works on output-capable pins so the test in the code seems to be in the wrong place;
- the `PULL_HOLD` behaviour documented with regard to pull-ups in deep-sleep seems to already be the default;
- `PULL_HOLD` actually results in output locking behaviour that is not documented;
- its only current connection to deep-sleep is that the GPIO will still be locked on wake;
- *actually* retaining drive/pull status during deep-sleep (/wake) requires use of a function that is not currently exposed;
- if one *does* enable hold-during-deep-sleep then the `PULL_HOLD` behaviour is actually the *reverse* of that documented in the quickref.

What would be the best changes here? Since `PULL_HOLD` doesn't currently do what it says it does, I'm curious as to whether anyone is actually using it.

I would have thought that this functionality is better moved to either some new `hold=True|False` keyword parameter or new `IN_HOLD`|`OUT_HOLD` pin modes?

Regardless, I'd definitely like to add a mechanism for calling `gpio_deep_sleep_hold_en()` to the ESP32 port. Honestly, the hold behaviour seems to be of such limited value without this that I wonder if it should just be enabled by default?

jonathanhogg, 2022-02-16:
Trying to understand the source of this "save power in deep-sleep with `PULL_HOLD`" meme, I've done some more digging. Inspired by the "of some pins" in the docs, I looked deeper into the Espressif docs and noticed [this](https://docs.espressif.com/projects/esp-idf/en/v4.3.2/esp32/api-reference/peripherals/gpio.html#_CPPv416rtc_gpio_isolate10gpio_num_t):

> `esp_err_t rtc_gpio_isolate(gpio_num_t gpio_num)`
>
> Helper function to disconnect internal circuits from an RTC IO This function disables input, output, pullup, pulldown, and enables hold feature for an RTC IO. Use this function if an RTC IO needs to be disconnected from internal circuits in deep sleep, to minimize leakage current.

Trying some tests I've been able to confirm that RTC-capable GPIO pins will by default retain their pull-up/-down status (though not output level) going into deep-sleep and waking back up from it. However, although this function enables pad hold, it is actually sufficient to simply disable the pull resistors with `pin.init(pull=None)` (and this will be retained on wake).

So would it be more correct for the docs to state that one should disable pull-ups/-downs for any RTC-capable GPIO pin (32, 33, 25, 26, 27, 14, 12, 13, 15, 2, 0, 4) where these might cause current to leak externally to the processor? Not all of these pins will be user-accessible, so this might somewhat depend on the module design.

As noted, for non-RTC-capable GPIO pins, the pull-up/-down resistors will already be disconnected by default on entering deep-sleep.

jonathanhogg, 2022-03-10:
More tests reveal that RTC pads will also retain their output level in deep-sleep if pad hold is enabled.

So, what I have in summary so far:

- pad *not held*:
  - RTC GPIO:
    - *retains* pull up/down through `deepsleep()`
    - *does not retain* output direction/level into `deepsleep()`
  - non-RTC GPIO:
    - *does not retain* pull up/down or output direction/level into `deepsleep()`
- pad *held*:
  - RTC GPIO:
    - *retains* pull up/down through `deepsleep()`
    - *retains* direction/output level going into `deepsleep()`, but not on waking
  - non-RTC GPIO:
    - *does not retain* pull up/down or output direction/level into `deepsleep()`
- pad *held* and `esp32.gpio_deep_sleep_hold(True)`:
  - non-RTC GPIO:
    - *retains* pull up/down or output direction/level into `deepsleep()`

Slight wrinkle in this is that the `VDD_SDIO` power domain pins, which includes GPIO pins 16 and 17, are powered down during deep-sleep (at least in the package I'm testing on). So while the configuration is retained for these pins with pad hold and `esp32.gpio_deep_sleep_hold(True)`, a high output level cannot be maintained.

Also, *all* pads are retaining their config through a `machine.reset()` and watchdog reset, so it seems I was wrong about that being a pad hold feature. I honestly can't make sense of the ESP32 documentation…

dpgeorge, 2022-03-21:
Thanks for the thorough investigation here.  The issue is now fixed by 7684c996bc2b2521fa4bd023e4fa24622092cd5f and 21d0599bd10aa567e3fdee5e39da1fe9a9f816e6

closed 2022-03-21 by dpgeorge