micropython/micropython #8289: ESP32: Static IP config crash with IDF >= 4.2

harbaum, 2022-02-10:
The following code

```
import network

station = network.WLAN(network.STA_IF)
station.active(True)
station.ifconfig( ( '192.168.0.2','255.255.255.0','192.168.0.1','8.8.8.8' ) )
```

works fine for me on MicroPython v1.18-133-ga57afa822 when  built with IDF 4.0.2 and 4.1.1

But when built with 4.2 and any later version i've tried the same code resets the board in the ifconfig line like so:

```
micropython/esp-idf/components/freertos/queue.c:718 (xQueueGenericSend)- assert failed!

abort() was called at PC 0x40096a53 on core 0

Backtrace:0x4009374e:0x3ffcde10 0x40093e95:0x3ffcde30 0x40097aa6:0x3ffcde50 0x40096a53:0x3ffcdec0 0x4013d512:0x3ffcdf00 0x4013d6bd:0x3ffcdf20 0x4013e2ed:0x3ffcdf40 0x4013ceab:0x3ffcdf70 0x4013ceca:0x3ffcdfa0 0x401e3d89:0x3ffcdfc0 0x401e3e4c:0x3ffce020

ELF file SHA256: 00b928da27c11fe5

Rebooting...
```

harbaum, 2022-02-10:
On 4.3.2 the crash also happens but looks a little different:

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400984ca  PS      : 0x00060f33  A0      : 0x80095eae  A1      : 0x3ffce330  
A2      : 0x0000004c  A3      : 0xffffffff  A4      : 0x3ffd70dc  A5      : 0x00000000  
A6      : 0x00000014  A7      : 0x3ffce550  A8      : 0x0000abab  A9      : 0x0000cdcd  
A10     : 0x0000abab  A11     : 0xb33fffff  A12     : 0x00060f20  A13     : 0x00060f23  
A14     : 0x000000a5  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000004c  LBEG    : 0x400e5a58  LEND    : 0x400e5a60  LCOUNT  : 0x00000000  

Backtrace:0x400984c7:0x3ffce330 0x40095eab:0x3ffce350 0x401482fa:0x3ffce390 0x401484a5:0x3ffce3b0 0x40147dcd:0x3ffce3d0 0x400fd9f7:0x3ffce3f0 0x400e1787:0x3ffce450 0x400e80b9:0x3ffce480 0x400e81e9:0x3ffce4a0 0x400845f9:0x3ffce4c0 0x400e1718:0x3ffce560 0x400e80b9:0x3ffce5c0 0x400e80e2:0x3ffce5e0 0x400f53fe:0x3ffce600 0x400f55fe:0x3ffce690 0x400da46a:0x3ffce6e0

ELF file SHA256: f8722c5795889d4c

Rebooting...
```

robert-hh, 2022-02-10:
I had these crashes as well, but only with 4.3 and up. And only if that code sequence was executed immediately in the startup phase, e.g. in main.py. Running that code a few seconds later works. Did not look into that yet.

robert-hh, 2022-02-10:
I re-tested that again. It seems to be a mis-conception. Sometimes it works, sometimes it does not. That has the faint smell of uninitialized data. It seems to work reliable if the IP setting is done after connect. But that is not the way I prefer to use static IP settings.

robert-hh, 2022-02-10:
platform.plazform() tells you next others the idf version.

harbaum, 2022-02-10:
Indeed, there is some race condition or the like. This code works on 4.3.2:

```
import network, time

station = network.WLAN(network.STA_IF)
station.active(True)
time.sleep(.1)
station.ifconfig( ( '192.168.0.2','255.255.255.0','192.168.0.1','8.8.8.8' ) )
```

robert-hh, 2022-02-10:
Of course, and in networks with DHCP that wastes the address pool. I do not mean it as a suitable approach, but more as a hint for bug hunting.

dpgeorge, 2022-02-12:
Official builds for standard esp32 (ie not S2, S3, C3 variants) are built with IDF v4.2.2.

jonathanhogg, 2022-02-17:
I've seen similar race-condition crashes calling `.config(dhcp_hostname=…)` immediately after `.active(True)` with IDF 4.3. Don't have a backtrace to hand, but I'm pretty sure it was a bad pointer somewhere in the IDF code.

Moving the rest of the WLAN config to happen slightly later in my boot process made the problem go away so I stopped digging further.

fliiiix, 2022-08-28:
I see this every time on the [v1.19.1 (2022-06-18) .bin relase](https://micropython.org/resources/firmware/esp32-20220618-v1.19.1.bin)

with this code:
```
import esp
esp.osdebug(0)

from network import WLAN, STA_IF

wlan = WLAN(STA_IF)
wlan.active(True)
wlan.ifconfig(("192.168.17.166", "255.255.255.0", "192.168.17.5", "192.168.17.6"))
wlan.connect("helene", "+facebook+")
print(wlan.ifconfig())
```

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12344
ho 0 tail 12 room 4
load:0x40080400,len:4124
entry 0x40080680
/home/micropython/esp-idf-v4.2/components/freertos/queue.c:733 (xQueueGenericSend)- assert failed!

abort() was called at PC 0x40096ed7 on core 0

Backtrace:0x40093b5e:0x3ffcdf50 0x400942d5:0x3ffcdf70 0x40097f26:0x3ffcdf90 0x40096ed7:0x3ffce000 0x4013ee56:0x3ffce040 0x4013f001:0x3ffce060 0x4013fc55:0x3ffce080 0x4013e7ef:0x3ffce0b0 0x4013e80e:0x3ffce0e0 0x401e4b39:0x3ffce100 0x401e4bfc:0x3ffce160
```

socketpair, 2022-09-14:
Same thing. `sleep(1)` after `sta_if.active(True)` helps!

jonathanhogg, 2022-09-14:
I haven't tried this in a while, so I just did a retest on my current build (4903e48e340bd9741577f30cacb31605cc70cf20 plus my two open PRs built on IDF v4.4.2).

Running:

```python
print("from network import *")
from network import *

print("wlan = WLAN(STA_IF)")
wlan = WLAN(STA_IF)

print("wlan.active(True)")
wlan.active(True)

print("wlan.ifconfig(('192.168.8.200', '255.255.255.0', '192.178.8.1', '192.168.8.1'))")
wlan.ifconfig(('192.168.8.200', '255.255.255.0', '192.178.8.1', '192.168.8.1'))
```

I get an instant explosion (backtrace elided for readability):

```
MicroPython v1.19.1-395-g868ddfc6f on 2022-09-14; ESP32 module with ESP32
Type "help()" for more information.
>>> import test
from network import *
wlan = WLAN(STA_IF)
wlan.active(True)
wlan.ifconfig(('192.168.8.200', '255.255.255.0', '192.178.8.1', '192.168.8.1'))

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4009638b  PS      : 0x00060a30  A0      : 0x8015cdc0  A1      : 0x3ffce3e0  
A2      : 0x00000000  A3      : 0x3ffd7e10  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80095e7c  A9      : 0x3ffce3a0  
A10     : 0x3ffc2d28  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000046  LBEG    : 0x400d32a0  LEND    : 0x400d32aa  LCOUNT  : 0x00000000  

0x40096388: vQueueDelete at .../esp-idf-4.4/components/freertos/queue.c:2137
0x4015cdbd: esp_netif_list_unlock at .../esp-idf-4.4/components/esp_netif/esp_netif_objects.c:60
 (inlined by) esp_netif_list_unlock at .../esp-idf-4.4/components/esp_netif/esp_netif_objects.c:55
0x4015cf50: esp_netif_get_handle_from_ifkey at .../esp-idf-4.4/components/esp_netif/esp_netif_objects.c:187
0x4015c881: netif_from_if at .../esp-idf-4.4/components/tcpip_adapter/tcpip_adapter_compat.c:76
 (inlined by) netif_from_if at .../esp-idf-4.4/components/tcpip_adapter/tcpip_adapter_compat.c:72
 (inlined by) tcpip_adapter_get_ip_info at .../esp-idf-4.4/components/tcpip_adapter/tcpip_adapter_compat.c:199
0x400fb7c3: esp_ifconfig at .../micropython/ports/esp32/modnetwork.c:153
0x400de703: fun_builtin_var_call at .../micropython/py/objfun.c:131
0x400e5435: mp_call_function_n_kw at .../micropython/py/runtime.c:693
0x400e5565: mp_call_method_n_kw at .../micropython/py/runtime.c:709
0x40084c21: mp_execute_bytecode at .../micropython/py/vm.c:1042
...
```

and trying a `.config()` alternative

```python
print("from network import *")
from network import *

print("wlan = WLAN(STA_IF)")
wlan = WLAN(STA_IF)

print("wlan.active(True)")
wlan.active(True)

print("wlan.config(dhcp_hostname='test')")
wlan.config(dhcp_hostname='test')
```

I get the crash:

```
MicroPython v1.19.1-395-g868ddfc6f on 2022-09-14; ESP32 module with ESP32
Type "help()" for more information.
>>> import test
from network import *
wlan = WLAN(STA_IF)
wlan.active(True)
wlan.config(dhcp_hostname='test')

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4000127a  PS      : 0x00060a30  A0      : 0x8015cf43  A1      : 0x3ffce3a0  
A2      : 0x3f41f324  A3      : 0x00000000  A4      : 0xfffffffd  A5      : 0x3ffce45b  
A6      : 0x000000ff  A7      : 0x00000000  A8      : 0x00000057  A9      : 0x3ffce350  
A10     : 0x00000001  A11     : 0x3ffd7fb8  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

0x4015cf40: esp_netif_get_handle_from_ifkey at .../esp-idf-4.4/components/esp_netif/esp_netif_objects.c:182
0x4015cb85: netif_from_if at .../esp-idf-4.4/components/tcpip_adapter/tcpip_adapter_compat.c:76
 (inlined by) netif_from_if at .../esp-idf-4.4/components/tcpip_adapter/tcpip_adapter_compat.c:72
 (inlined by) tcpip_adapter_set_hostname at .../esp-idf-4.4/components/tcpip_adapter/tcpip_adapter_compat.c:374
0x400fc6ed: network_wlan_config at .../micropython/ports/esp32/network_wlan.c:472
0x400de6f5: fun_builtin_var_call at .../micropython/py/objfun.c:126
 (inlined by) fun_builtin_var_call at .../micropython/py/objfun.c:112
0x400e5435: mp_call_function_n_kw at .../micropython/py/runtime.c:693
0x400e5565: mp_call_method_n_kw at .../micropython/py/runtime.c:709
0x40084c21: mp_execute_bytecode at .../micropython/py/vm.c:1042
...
```

So the common IDF path that is failing is `netif_from_if() -> esp_netif_get_handle_from_ifkey()`, which is inlined from both `tcpip_adapter_get_ip_info()` and `tcpip_adapter_set_hostname()`.

At this point, my understanding of the IDF networking implementation (and time to learn more) has been pretty much exhausted. However, I did take a quick gander at a couple of the IDF wifi examples and the order of setup and specific API functions being called seems to be quite different from the way `esp32/network_wlan.c` works.

jonathanhogg, 2022-09-20:
Related issues: #8792, #9236

TGiles1998, 2022-09-21:
I have also experienced this issue currently.

robert-hh, 2023-01-19:
Instead of the small delay after wlan.active() as mentioned in #9236, calling `wlan.ifconfig()` before` wlan.active(True)` works as well. But as a patch adding a delay in the implementation of wlan.active() seems better to me, because then one does not have to care about statement ordering.

dpgeorge, 2023-01-20:
This is hopefully fixed by d6bc34a13aa734d8b32e5768c021377ac4815029

closed 2023-01-20 by dpgeorge

dpgeorge, 2023-06-27:
A proper fix for this was done in bccbaa92b1fc6237f0f49a7f07cc194835fbf4e3