micropython/micropython #8290: ESP32: SDCard throws Errors

Narsskrarc, 2022-02-10:
Hi,

I try to access a SD card with MP V 1.18. The reader is connected to slot 2. The hardware seem to work (tested with an ESP32 with C code). It's a 8GB SDHC formatted with FAT32.

I do
`sd = SDCard(slot=2)`

Now
`sd.info()`
throws
`OSError: (-265, 'ESP_ERR_INVALID_CRC')`

and
`os.mount(sd, "/sd")`
throws
`OSError: 16`

16 seems to be EBUSY, although this errorcode isn't defined in errno.errorcode... and it doesn't help me either.

What can I do to investigate what goes wrong?

Thanks.

Narsskrarc, 2022-02-12:
I investigated a bit further on the first part - `sd.info()`.
I listened what happens with a logic analyzer. Anything seems to be ok. The last command I've seen is a ACMD51 (SEND_SCR); I checked the CRC and it's correct.
Perhaps it stumbles over the nine 0xFF before the start block token (0xFE).

But I can't find the corresponding souce... I think the source is in `ports/esp32/machine_sdcard.c`. There's an `#include "driver/sdmmc_host.h"` and other includes inside which I can't locate anywhere in the micropython sourcecode.

Narsskrarc, 2022-02-13:
Now I did the same for `os.mount()`.
It is exactly the same sequence as with `sd.info()`... it stops with ACMD51.

So it seems the reason doesn't lie in the communication with the card, but in the SDCard class itself. As mentioned I'm not able to dive deeper in the source code, because I simply haven't found it.

Josverl, 2022-02-14:
`driver/sdmmc_host.h` is part of the esp-idf for the esp32.
assuming you have cloned the repo you can setup to toolchain by running:
``` bash
source tools/ci.sh
ci_esp32_idf402_setup
```
If you use codespaces then you can even run that in your browser to allow you compare side by side.

Narsskrarc, 2022-02-15:
OK, that's why I didn't find it.
Since the SPI communication seems to work I don't think it's a problem from the SPI implementation; wether I have to search in the SDCard implementation or elsewhere I don't know yet.

I'm using a pure python implementation now (esp_sdcard-py). And since my quarantine is over now, I won't have much time to look into that in the near future...

Perhaps someone out there who knows these errors already?

Narsskrarc, 2022-02-16:
I browsed a bit in the IDF sources now, but I did not even find the source of the EBUSY error.

Is there anybody out there who is using machine.SDCard with esp32 successfully?

Josverl, 2022-02-16:
In to help out further you will need to share addtl info on your exact hardware setup, and on the code you use (try ) to initialize your card. 
I have many esp32 boards with SD cards that work in micropython, however there is a wide variety in how they are connected , to which pins , and with which data lines , and hence in the initialization needed.

Narsskrarc, 2022-02-16:
> perhaps https://forums.micropython.org is a better place for your ask.

Hm, another forum... there are so many forums I have to watch... but I'll give it a try ;-)

Narsskrarc, 2022-02-28:
Well, Mike from the micropython forum reminded me that machine.SDCard works with 20MHz, if you don't change that with freq=... so I tried looked again at the spi bus with the full 24MHz of my little logic analyzer. And, yes, there was a little glitch which turned out to be a unsuccessful try of communication. I must have overseen that in my previous attempts.

When I use freq=... it works in fact. But no matter which frequency I use it uses a transfer rate of 400kHz (tried with 19MHz, 10MHz, 1.32MHz, 1Mhz).

Any ideas why?

jonnor, 2024-09-08:
@Narsskrarc did you manage to resolve your issue? Is it still present in recent MicroPython versions, like 1.23 ?

Narsskrarc, 2024-09-08:
Well, no, I didn't test in the last time. Sadly, the project's priority has went very low the last time...
The last thing I did was pimping sdcard.py a bit (see [here](https://forum.micropython.org/viewtopic.php?f=18&t=12007#p65611)). That worked, but I don't know if there are working versions out of the box now.