micropython/micropython #8292: Adding support for STM32G0 MCUs

AsensioL, 2022-02-11:
Hi I have been working on adding support for STM32G0 MCUs. Particularly, the ST NUCLEO_G0B1RE for a couple of weeks, and today I got it to compile, link and run finally. I was also able run the run-tests.py script (see results below).
This is my first contribution to a project.

This requires the stm32lib repository to be updated to include HAL and CMSIS for the STM32G0.

AsensioL, 2022-02-11:
What is the best way to share my changes?
Should I fork and push them there?
The contributing guidelines and the git history show small and concise changes on each commit, and this would require several changes to around 20 files

AsensioL, 2022-02-12:
Thanks, I'll start working on that.

AsensioL, 2022-02-12:
I just created branch: 
I pulled from master, and resolving conflicts right now. I also noticed MCU families tend to be grouped alphabetically, so I'm making that way too.

AsensioL, 2022-02-14:
While porting the code for the G0, on powerctrl.c...
- I used the SR register for the wake up flags (WUTF...), which looked similar to the STM32H7A3xx STM32H7B3xx port ([link](https://github.com/AsensioL/micropython/commit/f674d53c84cb2d3b9e67958c2dacd6a8fbb74ea3#r66615271))
- But I noticed that, on the G4/WL port, the MISR register is used. Is that right?

Since I'm not sure, I'm going to merge the code for the G0 like the G4/WL, but I thougth I should mention.

AsensioL, 2022-02-14:
Pulled from master and conflicts resolved.
Waiting for feedback now.
https://github.com/AsensioL/micropython/tree/support_stm32g0

dpgeorge, 2022-02-14:
Please open a pull request from that branch to this repo.

dpgeorge, 2022-04-28:
Done in 9d08eb02474fcace0a81dd3518b938afbf1486eb

closed 2022-04-28 by dpgeorge

alxkos, 2025-09-30:
hey, is it working well? all the components? thanks