micropython/micropython #8293: Support datetime for ESP32 (maybe others)

chrisdecker1201, 2022-02-11:
Hello,

I'm needing a subset of the CPython `datetime` library on my ESP32.

* Creating `datetime` & `timedelta` objects
* calculating with `datetime` & `timedelta` objects
* compare `datetime` & `timedelta` objects

I've seen that you already provide this functionality for the [unix-port](https://github.com/micropython/micropython-lib/tree/master/unix-ffi/datetime).

As I implement this anyway, I would ask if I should/could integrate this into micropython.

I think it's not necessary to have this functionality on every board, but maybe there are some ideas about this functionality already.

peterhinch, 2022-02-11:
See https://github.com/micropython/micropython-lib/pull/449

chrisdecker1201, 2022-02-11:
Thank you! That helps me a lot!