micropython/micropython #8294: ESP32 Generic: ESP-IDF v4.4 throws exception allocation errors (`MemoryError: memory allocation failed`)

RonnyLV, 2022-02-11:
Building ESP-32 GENERIC board using ESP-IDF v4.4 (`142bb32c50fa9875b8b69fa539a2d59559460d72`) throws memory allocation errors upon py file imports:

```
Traceback (most recent call last):
  File "main.py", line 62, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "main.py", line 38, in setup
  File "./sargs.py", line 70, in setup
  File "./sargs.py", line 80, in _init_lcd
  File "./ssd1306.py", line 113, in __init__
MemoryError: memory allocation failed, allocating 1025 bytes
MicroPython 11ab899 on 2022-02-10; ESP32 module with ESP32
Type "help()" for more information.
```

Cannot reproduce with ESP-IDF v4.0.2 and also with v4.2.2 everything is fine.

dpgeorge, 2022-02-11:
IDF 4.4 (and I think 4.3) has a more fragmented IDF heap, and so there is less heap available for MicroPython.  On IDF 4.0.2 and 4.2.2 there is about 100k available for MicroPython, but on 4.4 it's only 64000.

See related #7214.

wsyfree, 2022-02-15:
> 

Is this the reason why MPY cannot adapt to idf4 4 ？  Does it need to be repaired by the espressif official?

wsyfree, 2022-02-15:
Related issue #8277 #8279 #8183