micropython/micropython #8295: Arduino pins aliases for Nucleo-WB55 ?

YannickMarietti, 2022-02-11:
Hello

I'am not skilled enough with GiHub, I suppose my post would better fit in a "Pull request". Sorry :(

**My point is the following :** 
Nucleo WB55 board (as all Nucleo 64 boards) have Arduino standard connectors. So I think it would make sense to add inside the _pins.csv_ file the aliases for Arduino pins names.

If we look specifically at the Nucleo-WB55 "pins.csv" I propose to change its content this way:

```
A3,PA0
A2,PA1
D1,PA2
D0,PA3
D10,PA4
D13,PA5
D12,PA6
D11,PA7
D6,PA8
D9,PA9
D3,PA10
,PA11
,PA12
,PA13
,PA14
D5,PA15
,PB0
,PB1
,PB2
,PB3
,PB4
,PB5
,PB6
,PB7
D15,PB8
D14,PB9
,PB10
,PB11
,PB12
,PB13
,PB14
,PB15
A0,PC0
A1,PC1
A5,PC2
A4,PC3
,PC4
,PC5
D2,PC6
D4,PC10
,PC11
D8,PC12
D7,PC13
,PD0
,PD1
,PE4
SW,PC4
SW1,PC4
SW2,PD0
SW3,PD1
LED_GREEN,PB0
LED_RED,PB1
LED_BLUE,PB5
USB_DM,PA11
USB_DP,PA12
```

Best regards

Yannick

robert-hh, 2022-02-11:
pins.csv it just an alias list, assigning names to pins. So you have to add entries only for the pins you want to give a name. And this list does not have to be sorted by pin numbers. So you can sort it by alias names. A pin can have more than one alias.

YannickMarietti, 2022-02-12:
Thank you very much for this clarification.
But is it possible to include these Arduino aliases in futures firmwares builds for Nucleo-WB55 ?

mattytrentini, 2022-08-14:
This ticket can also be closed now @dpgeorge

closed 2022-08-15 by dpgeorge