micropython/micropython #8296: ESP32: I2S how to stream uninterrupted with uasyncio?

Narsskrarc, 2022-02-13:
Hi,

I try to play a wav file with a coro like that:

    async def play(self, vol):
        vshft = (vol // 10) - 10
        audio_out = I2S(0,
                        sck = Pin(I2S_BCK), ws = Pin(I2S_LRC), sd = Pin(I2S_DOUT),
                        mode = I2S.TX,
                        bits = 16,
                        format = I2S.MONO,
                        rate = 16000,
                        ibuf = 20000)
        
        print("Playback start {}".format(self.file))
        
        wav = open(self.file, 'rb')
        pos = wav.seek(44)
    
        BUF_SIZE = 4096
        wav_samples = bytearray(BUF_SIZE)
        wav_samples_mv = memoryview(wav_samples)
    
        try:
            audio_writer = asyncio.StreamWriter(audio_out)
            num_read = wav.readinto(wav_samples_mv)
            I2S.shift(buf=wav_samples_mv, bits=16, shift=vshft)
            while num_read > 0:
                if num_read < BUF_SIZE:
                    audio_writer.write(wav_samples_mv[0:num_read])
                else:
                    audio_writer.write(wav_samples_mv)
                num_read = wav.readinto(wav_samples_mv)
                I2S.shift(buf=wav_samples_mv, bits=16, shift=vshft)
                await audio_writer.drain()
        except (KeyboardInterrupt, Exception, OSError) as e:
            print("caught exception {} {}".format(type(e).__name__, e))
            raise
        finally:
            wav.close()
            audio_writer.close()
            audio_out.deinit()
            print("Playback stop")

The coro works as expected on the whole. Though it is a bit crackling.
I would think it would be better to fill up the I2S buffer **before** draining it; but there's no possibility to do that. If I leave the drain() out, the buffer overruns at no time (not unexpected).

There should be a possibility to set the 'low watermark', at which drain() returns.

Or perhaps there is a better solution to be sure the buffer doesn't run dry?

peterhinch, 2022-02-13:
See [this CD quality player demo](https://github.com/peterhinch/micropython-micro-gui/blob/main/gui/demos/audio.py). Unfortunately it's not documented, but the audio playing has to share processing with the display refresh. The key routine [is here](https://github.com/peterhinch/micropython-micro-gui/blob/01b00c5e2c0b455f7f2a19e96e267b28c2efd522/gui/demos/audio.py#L235).

Narsskrarc, 2022-02-14:
Thank you!
That's in fact more compact and I've put the initialization of I2S in the class __init__.
It's not perfect yet - the SD handling seems to be too slow for 44100/stereo. 22050/mono works, but still a bit crackling.

And there's a short noise (about 0.2 sec) after playback... I don't know why yet.

robert-hh, 2022-02-14:
I found wiring an external I2S module quite critical. The wires should be as short as possible. And still touching or moving caused crackling. For the mimxrt board a solution was to increase the impedance of the GPIO driver. That may not be possible or foreseen in the ESP32 port. But if wire sensitivity is a problem, you could try inserting resistors into the wire as close as possible to the ESP32. A good range is 50-100 Ohm.

Narsskrarc, 2022-02-14:
> I found wiring an external I2S module quite critical. The wires should be as short as possible. And still touching or moving caused crackling.

Well, at the moment it resides on a breadboard... so I can shorten the wires later.
> For the mimxrt board a solution was to increase the impedance of the GPIO driver. That may not be possible or foreseen in the ESP32 port. But if wire sensitivity is a problem, you could try inserting resistors into the wire as close as possible to the ESP32. A good range is 50-100 Ohm.

On all three wires?

robert-hh, 2022-02-14:
> On all three wires?

Generally yes. You could of course try to do it in the Bit clock first.

Narsskrarc, 2022-02-14:
Yes, that's better indeed. Thank you!

peterhinch, 2022-02-14:
I'll second that comment about wiring. I couldn't get it to work on a breadboard and ended up designing a PCB with very short traces.

miketeachman, 2022-02-15:
> Or perhaps there is a better solution to be sure the buffer doesn't run dry?

This is unlikely to be a problem.  At 44100, the time to send a 4096 byte buffer of samples to the I2S hardware DAC is about 5x longer than the time to read 4096 bytes from an SD card on the ESP32 (assuming a decent quality SD card).  If this happens, perhaps revisit your asyncio design, or do an extra read/write before entering the while loop.

> And there's a short noise (about 0.2 sec) after playback... I don't know why yet.

Did you check that the data section of the Wav file starts at byte 44?  There can often be copyright, author, title, etc information before the data section.  If that is the case, byte 44 could be in the middle of something like a song title, and your program will start by sending non-audio data to the I2S DAC, resulting in noise.

You can investigate this theory with an online hex editor such as:  https://hexed.it/

If the data section does not start at byte 44, take a look at the parse() method in this code:  
https://github.com/miketeachman/micropython-i2s-examples/blob/master/examples/wavplayer.py

miketeachman, 2022-02-15:
> Well, at the moment it resides on a breadboard... so I can shorten the wires later.

A soldered solution can be low-cost, low-effort, and effective.  For example, use Adafruit's perma-proto as a protoboard. Plug the development board and I2S DAC board into soldered-in headers.  Use soldered-in 22 AWG wire to make the point-to-point connections.  Keep the wires as short as possible.  With this approach I have never had a problem with noise.

Narsskrarc, 2022-02-15:
> This is unlikely to be a problem. At 44100, the time to send a 4096 byte buffer of samples to the I2S hardware DAC is about 5x longer than the time to read 4096 bytes from an SD card on the ESP32 [...]

Since the machine.SDCard doesn't work with me (open get's an EBUSY error, see other thread), I'm using esp_sdcard.py as a python-only implementation. It's a bit slower, so I can't get above 18MHz. I'll see where I can optimize here.

> Did you check that the data section of the Wav file starts at byte 44? There can often be copyright, author, title, etc information before the data section. If that is the case, byte 44 could be in [...]

Oh, thanks, I'll look into that. But the noise comes **after** the playback, not before, so that's not the problem here.

Narsskrarc, 2022-02-15:
> A soldered solution can be low-cost, low-effort, and effective. For example, use Adafruit's perma-proto as a protoboard. Plug the development board and I2S DAC board into soldered-in headers. Use [...]

I planned to make a soldered version anyway, so I will have a look at the wire lengths then. Usually I'm using a perfboard for that, so I can place the components as I wish.

Narsskrarc, 2022-02-15:
I think the main problem is solved, with a bit better software and the hints to the wire length.
So I'll close it, though I don't have a clue about the trailing noise yet.

closed 2022-02-15 by Narsskrarc

robert-hh, 2022-02-15:
One comment, if you use the sdcard.py driver: This driver uses a pretty slow SPI baud rate. You could try to increase it. It is in function init_card() around line 114. It sets the SPI baud rate to 1_320_000. But it can as well be much higher, increasing the transfer rate. On a MIMXRT board, I used 40MHz.

Narsskrarc, 2022-02-15:
Oh yes, I saw it. But with 20MHz on my ESP32 it doesn't find the SD any more; so this seems to be the maximum.

miketeachman, 2022-02-15:
I found that the SPI speed can be increased outside the driver, as shown in this example, for the raspberry pi pico:
https://github.com/miketeachman/micropython-i2s-examples/blob/master/examples/play_wav_from_sdcard_blocking.py
Look for "# this has no effect on spi bus speed to SD Card"
I checked that this worked using an oscilloscope.
It would be worth trying to understand why the native machine.sdcard() class does not work.  It is so much faster than the sdcard.py driver.

Narsskrarc, 2022-02-15:
Well, this comment is not correct as far as I can see. The initialization usually works with a lower speed (e.g. 100_000), but for the data transfer it switches to the baudrate which is set in the constructor. It should depend on what the sdcard can manage, but that seems not to be implementated.

robert-hh, 2022-02-15:
> but for the data transfer it switches to the baudrate which is set in the constructor.

I just changed the code from drivers/sdcard/sdcard.py that way. Which version do you use?

Narsskrarc, 2022-02-15:
Hm... this one:
[https://github.com/micropython/micropython/issues/7932](https://github.com/micropython/micropython/issues/7932)

miketeachman, 2022-02-16:
> but for the data transfer it switches to the baudrate which is set in the constructor.

Hmmm ... I wonder if we are looking at the same sdcard.py.  I am looking at the official MicroPython sdcard driver.
https://github.com/micropython/micropython/blob/master/drivers/sdcard/sdcard.py

Narsskrarc, 2022-02-16:
No, it's an enhancement from saul530 (see above). He changed initializing to make re-initializing more robust. And switches to the requested baudrate after initialization. ;-)

Narsskrarc, 2022-02-16:
> It would be worth trying to understand why the native machine.sdcard() class does not work. It is so much faster than the sdcard.py driver.

Yes, it surely would. I looked into the sources again yesterday, but I don't even find out where the error is generated... that'll be much work I think.

BTW: Am I the only one who tries to use machine.SDCard with esp32? Or does it work elsewhere?

robert-hh, 2022-02-16:
I made a PR yesterday to include the Baudrate argument to the constructor. https://github.com/micropython/micropython/pull/8307

miketeachman, 2022-02-20:
> BTW: Am I the only one who tries to use machine.SDCard with esp32? Or does it work elsewhere?

For the ESP32, I only use the machine.SDCard class and so far I have encountered no problems.  Today, I programmed an ESP32 (Lolin D32 Pro with a built-in SD card slot) with the latest v1.18 version.  I ran a read/write SD card test, using a Class 4 and a Class 10 SD card.   I did not see any failures or exceptions.  This does not help you to find a root cause, but at least you know that the SDCard class can work on some hardware and SD cards.

miketeachman, 2022-02-20:
> Oh, thanks, I'll look into that. But the noise comes **after** the playback, not before, so that's not the problem here.

Sorry, I misunderstood your description.  If you have time, I would like to investigate this noise issue.  Perhaps there is a bug in the machine.I2S implementation that I have missed.

Is it possible to provide a minimal source code example that demonstrates the post-playback noise problem?

Are you using the v1.18 MicroPython version?  I imagine you are, but I did not see that mentioned in the comments.

Narsskrarc, 2022-02-20:
> For the ESP32, I only use the machine.SDCard class and so far I have encountered no problems. Today, I programmed an ESP32 (Lolin D32 Pro with a built-in SD card slot) with the latest v1.18 version. [...]

Well, that helps indeed... so I know that I have to investigate further. I only have Dev Kit C, but other sd cards to test.

Narsskrarc, 2022-02-20:
> Sorry, I misunderstood your description. If you have time, I would like to investigate this noise issue. Perhaps there is a bug in the machine.I2S implementation that I have missed.
>
> Is it possible [...]

I'll try. But it could take a time.

> Are you using the v1.18 MicroPython version? I imagine you are, but I did not see that mentioned in the comments.

Yes, it's 1.18.

miketeachman, 2022-02-21:
Sounds good.