micropython/micropython #8300: ESP32: os.listdir with non-ASCII chars

Narsskrarc, 2022-02-14:
Hi,

I have a file on the SD-Card names "Frühstück.wav". With `os.listdir` it is displayed as "Frhstck.wav", When I convert it to `bytes()` I can see it is b"Fr\x81hst\x81ck.wav" in fact.

I have to compare the name to a string named "Frühstück.wav". But I can't convert neither the string from os.listdir in unicode nor can I create a string which is equal to the string from os.listdir.
I found the hint to use a unicode directory for os.listdir(). I tried to decode(anything) the b"...", but it throws an UnicodeError no matter which encoding I try. The other way round I can encode the unicode string with any encoding, it gets the same byte string:
`"Frühstück.wav".encode('UTF-8') -> b'Fr\xc3\xbchst\xc3\xbcck.wav'`
`"Frühstück.wav".encode('ASCII') -> b'Fr\xc3\xbchst\xc3\xbcck.wav'`
`"Frühstück.wav".encode('ISO-8859-1') -> b'Fr\xc3\xbchst\xc3\xbcck.wav'`

What is the right way to handle non-ascii-filenames with micropython?

smurfix, 2026-01-29:
File contents are OK to raise annoying exceptions when it's not UTF-8. That's what binary mode is for.

My problem with error="replace" is that it's not reversible, thus I still can't delete that file. The "surrogateescape" method (using `U+DC80..U+DCFF` to represent non-UTF8-ish bytes; see also https://peps.python.org/pep-0383/) would be more helpful.

smurfix, 2026-01-30:
You're correct -- file names are a sequence of bytes, and most file systems don't care about their meaning. The only forbidden values are null and slash.

There still are legacy file systems with non-UTF8 filenames out there.