micropython/micropython #8301: uasyncio tests can fail on windows ports

stinos, 2022-02-14:
Since #8030 the dev variant yields 'random' test failures for uasyncio, probably depending on machine/version/...

For example on my machine the uasyncio_basic consistently prints `took 30 50 0` as last line, instead of the expected `took 20 40 0`. Some builds of the mingw port also fail in Appveyor for me. The reason for this is mentioned in #7796 :

> most likely all sleep/wait implementations on windows by default use a 16mSec (or close to that) resolution

The dev variant enables both uasyncio and the scheduler, and uasyncio calls `poll_poll_internal` and that uses `MICROPY_EVENT_POLL_HOOK` which in turn calls `mp_hal_delay_us(500)`. That last one is the culprit.

Setting `MICROPY_ENABLE_SCHEDULER` to 0 leaves `MICROPY_EVENT_POLL_HOOK` undefined, so doesn't compile because moduselect.c uses that unconditionally. Just adding `#define MICROPY_EVENT_POLL_HOOK` fixes that. It also makes the uasyncio tests pass. But it also turns `poll_poll_internal` into a busy loop. Likewise for `mp_hal_delay_ms` because it's conditional on `ifdef MICROPY_EVENT_POLL_HOOK` which should argueably should be `#if MICROPY_ENABLE_SCHEDULER`.

Things also grew a bit messy over time with there being multiple ways to sleep on Windows: `mp_hal_delay_ms` calls `SleepEx` but `mp_hal_delay_us` uses an implementation which uses which uses a timer. Which I wrote, but I don't remember why: I don't think it's better/worse/very different from SleepEx. Could be that things used to be different back then. Anyway these 2 should be consolidated. 

Then the question remains how to fix this
- I don't hink having busy waits is ok, even though they will make everything work and do so fairly accurately.
- by default there's just no accurate wait on windows. We could work with that, add a word of warning to the README, and change the tests so they pass
- figure out a way to make the wait accurate. I know at least one but it's generally considered bad practice (because https://randomascii.wordpress.com/2013/07/08/windows-timer-resolution-megawatts-wasted/). There might be other ways though, not sure.

stinos, 2022-03-01:
I tested some things and the Appveyor test failures are remedied by increasing the sleep periods (mostly using >100mSec instead of the typical 20 to 50mSec used now, to account for the default 15.6mSec resolution windows uses) in the tests I've seen failing. This does mean tests run longer but that's not really an issue I think.

The other option (bascially increasing the kernel timer interrupt rate, which affects the whole system) also works and makes the tests pass. It's convenient in the sense that it makes the windows port suitable for testing asyncio with small wait periods.

I'm inclined to implement the first option, and add a section in the readme which shows how to create a variant which implements the second. @dpgeorge what do you think?

dlech, 2022-03-30:
It would be more work are require a separate build, but a more deterministic way to run tests that depend on the passage of time is to create a "fake" clock that doesn't use the system clock at all. Instead, the clock state is driven programatically. This means tests that use this would have to be modified to tell the clock to tick X times, then test a condition, tick again, test something else, etc... But it eliminates the intermittent failures that inevitably come with relying on a real clock.

stinos, 2022-03-31:
> And it might even get these timing tests to pass on mac (which has similar issues with time resolution).

Wasn't aware of this, but makes sense.

> Another alternative is to have a build (or command line option) that runs the clock at a rate * R but still report "normal" ticks to the program

This is essentially the same as slowing down the tests, which, I found out today, still isn't sufficient to get them to always pass on windows: all my 100mSec sleeps now actually take 110mSec (which wasn't the case earlier). Combined with the overall test time increase, I'm thinking it's time to give up on this idea.

> It would be more work are require a separate build, but a more deterministic way to run tests that depend on the passage of time is to create a "fake" clock that doesn't use the system clock at all. Instead, the clock state is driven programatically. 

That's what I usually do: have a single swappable time source from which everything else derives (this is great e.g. when the entire application runs on top of streaming data from an ADC and you use each sample as a tick, and then for the tests that just gets swapped for a fake timer to which the actual time gets set in the tests, very easy and versatile - as long as you don't need 'real clock' time somewhere).

> This means tests that use this would have to be modified to tell the clock to tick X times

Not necessarily, or at least: not explicitly. For the type of tests MicroPython has it should be suffcient to make all `tick` functions increase a uSec tick for instance (to make sure there are no infinite polling loops), and then have the `sleep` function increase that tick and round to the next mSec (so that a sequence of tick/sleep(10)/tick/tick_diff still returns 10). I quickly tested this for the windows port and even with a pretty dumb implementation it does pass all tests.
There's still mod_time_sleep() in unix/modtime.c which uses plain `select` (but only when MICROPY_PY_BUILTINS_FLOAT which looks like an oversight). Not sure what to do there: get rid of the select and use sleep instead, or perhaps implement it via moduselect?
Would it be worth continuing this path?

Another alternative: change all timing tests to detect if they run on the unix/windows port and if so change all equality comparisons to not be exact but in a range like +-10mSec of the expected value. There aren't that many tests so should be fine.

dpgeorge, 2022-03-31:
> This is essentially the same as slowing down the tests, which, I found out today, still isn't sufficient to get them to always pass on windows: all my 100mSec sleeps now actually take 110mSec (which wasn't the case earlier).

I'm surprised this doesn't work.  The tests already have `round(dt, -1)` for the check on the timing, which is essentially +/- 10ms error margin.  Doing a 10x slow down on the timing of the test should give 10 time better accuracy on the dt measurement, which should be within those error margins.

Yet another option: change the tests so they don't test timing, just ordering of events.  Then have a few special tests that do test timing, one set which tests rough timing (for windows/mac/etc) and another that tests accurate timing (for bare metal).

stinos, 2022-05-03:
> The tests already have round(dt, -1) for the check on the timing, which is essentially +/- 10ms error margin. 

That's actually +-5mSec but that's covered in #8617 already :)

dpgeorge, 2022-05-03:
> That's actually +-5mSec

You are right!  So changing it to `round(dt, -2)` will make it 50ms which seems good enough.

dpgeorge, 2022-05-04:
Tests timing was increased in c90f097519c7342a128848bf9dc5b38ba75d4222.

Windows sleep was improved/cleaned up in 8aa79c95bd3960763c94e0ea9e43e2e6bfd20a74 and 20028c7c807d91a1d8ce2195580005068112f9a5

PR #8617 (2022-05-04): tests/extmod: Increase timing on uasyncio tests to make more reliable.
This aims to fix #8301, where uasyncio tests can fail on Windows.  But they also fail sometimes on Linux, and also regularly on macOS.  So increasing the timing seems to be a reasonable solution.
closed 2022-05-04 by dpgeorge