micropython/micropython #8308: ESP32-S3 REPL error - ESP32-S3-DEVKITC-1-N8R2

beyonlo, 2022-02-15:
Hi all.

### Operating System

Ubuntu 20.4

### Module Description

Chip is ESP32-S3 - module ESP32-S3-WROOM-1 (ESP32-S3R2)

### Kit Description

PartNumber: ESP32-S3-DEVKITC-1-N8R2 -> ESP32-S3 -DevKitC-1 v1.6 from espressif using the module ESP32-S3-WROOM-1 (ESP32-S3R2)

https://www.digikey.com/en/products/detail/espressif-systems/ESP32-S3-DEVKITC-1-N8R2/15199627

**I have REPL error on that kit.**
PS: I tested REPL using the UART and USB on KIT and no success.

The erase flash and write flash with MicroPython 1.18 works all fine, but I can't to access the REPL. I tested via picocom and mpremote. I tried to write a python file using the ampy and nothing.

**Is that a bug or Am I doing something wrong?**

 **All tests and process are bellow:**

```

pi@pi:~/esp32/esp32-s3$ ls -la
total 1232
drwxrwxr-x 2 pi pi    4096 fev 15 15:54 .
drwxrwxr-x 3 pi pi    4096 fev 15 12:57 ..
-rw-rw-r-- 1 pi pi 1246976 jan 17 00:09 GENERIC_S3_SPIRAM-20220117-v1.18.bin
-rw-rw-r-- 1 pi pi      12 fev 15 15:54 test.py
pi@pi:~/esp32/esp32-s3$ 
pi@pi:~/esp32/esp32-s3$ ls -l /dev/ttyACM0 
crw-rw---- 1 root dialout 166, 0 fev 15 15:55 /dev/ttyACM0
pi@pi:~/esp32/esp32-s3$ 
pi@pi:~/esp32/esp32-s3$ sudo esptool.py --chip esp32s3 --port /dev/ttyACM0 erase_flash
esptool.py v3.2
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e0:65:70
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.0s
Hard resetting via RTS pin...
pi@pi:~/esp32/esp32-s3$ sudo esptool.py --chip esp32s3 --port /dev/ttyACM0  write_flash -z 0x1000 GENERIC_S3_SPIRAM-20220117-v1.18.bin 
esptool.py v3.2
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e0:65:70
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00131fff...
Compressed 1246976 bytes to 815997...
Wrote 1246976 bytes (815997 compressed) at 0x00001000 in 13.7 seconds (effective 725.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
pi@pi:~/esp32/esp32-s3$ 
pi@pi:~/esp32/esp32-s3$ 
pi@pi:~/esp32/esp32-s3$ sudo picocom /dev/ttyACM0 -b115200
picocom v3.1

port is        : /dev/ttyACM0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

FATAL: cannot open /dev/ttyACM0: No such file or directory
pi@pi:~/esp32/esp32-s3$ mpremote 
no device found
pi@pi:~/esp32/esp32-s3$ sudo ampy -p /dev/ttyACM0 put 
GENERIC_S3_SPIRAM-20220117-v1.18.bin  test.py                               
pi@pi:~/esp32/esp32-s3$ sudo ampy -p /dev/ttyACM0 put test.py 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 537, in write
    n = os.write(self.fd, d)
OSError: [Errno 19] No such device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 8, in <module>
    sys.exit(cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/ampy/cli.py", line 264, in put
    board_files.put(remote, infile.read())
  File "/usr/local/lib/python3.8/dist-packages/ampy/files.py", line 213, in put
    self._pyboard.enter_raw_repl()
  File "/usr/local/lib/python3.8/dist-packages/ampy/pyboard.py", line 181, in enter_raw_repl
    self.serial.write(b'\r\x03')
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 571, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 19] No such device
pi@pi:~/esp32/esp32-s3$ 
pi@pi:~/esp32/esp32-s3$ 
pi@pi:~/esp32/esp32-s3$ 
pi@pi:~/esp32/esp32-s3$ ls -s /dev/ttyUSB0 
0 /dev/ttyUSB0
pi@pi:~/esp32/esp32-s3$ sudo picocom /dev/ttyU^C
pi@pi:~/esp32/esp32-s3$ sudo picocom /dev/ttyUSB0 -b115200
picocom v3.1

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready
d heq�B������ 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
�ESP-ROM:esp32s3-20210327f
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048836
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
```

beyonlo, 2022-02-16:
Anyone can tell me what exactly I need to do to works?

Thanks.

dpgeorge, 2022-02-17:
I do not have an ESP32-S3 to test with.

Try using the GENERIC_S3 firmware, to see if the SPIRAM is the issue.

stephanelsmith, 2022-02-19:
I may be wrong, but I think I remember getting this when I don't have a firmware actually loaded.  Unlike other esp32 versions, I found for S3, you need to use the flash command the build/make process provides when completed, not the generic/simple one as you've done. 

```
.....build.....
Project build complete. To flash, run this command:
xxxxx/idf4.4_py3.9_env/bin/python ../../../esp-idf-4.4rc/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after no_reset --chip esp32s3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 xxx/bootloader/bootloader.bin 0x8000 xxx/partition_table/partition-table.bin 0xd000 xxx/ota_data_initial.bin 0x10000 xxx/micropython.bin
```

rale, 2022-02-21:
Just flashed one of these.  The offset (-z 0x1000) is wrong, and should be 0 instead.

`esptool.py --baud 460800 write_flash -z 0 .\GENERIC_S3-20220117-v1.18.bin`

beyonlo, 2022-02-21:
> Just flashed one of these. The offset (-z 0x1000) is wrong, and should be 0 instead.
> 
> `esptool.py --baud 460800 write_flash -z 0 .\GENERIC_S3-20220117-v1.18.bin`

Thank so much @rale, that works perfectly.

@dpgeorge or @jimmo or other people that has permission to change the website docs: could you please, change to correct `offset doc` for the ESP32-S3? 

This url https://micropython.org/download/GENERIC_S3_SPIRAM/ and this https://micropython.org/download/GENERIC_S3/ 

Need just to change from `-z 0x1000` to `-z 0` to flash the MicroPython firmware for the S3. 

Thank you!

dpgeorge, 2022-02-25:
The update to the website for the programming offset has been fixed in 546e213265872cc6393af83a94bb2af5092d4348