micropython/micropython #8321: Arduino RP2040 nano connect - Microdot

JohannesMaierhofer, 2022-02-17:
I'm on Arduino RP2040 connect with Micropython 1.18
I would like to run [Microdot](https://github.com/miguelgrinberg/microdot) in combination with Wifi AP.
I could start the server, but the first request (using curl) causes the following issue: 

The minimum example-code is:
```
import network

SSID ='My_Nano_RP2040_Connect'   # Network SSID
KEY  ='1234567890'  # Network key (must be 10 chars)

wlan = network.WLAN(network.AP_IF)
wlan.active(True)
wlan.config(essid=SSID, key=KEY, security=wlan.WEP, channel=2)
print("AP mode started. SSID: {} IP: {}".format(SSID, wlan.ifconfig()[0]))

from microdot import Microdot

app = Microdot()

@app.route('/')
def hello(request):
    return 'Hello, world!'

app.run(port=80, debug=True)
```

```
Starting threaded server on 0.0.0.0:80...
Traceback (most recent call last):
  File "<stdin>", line 21, in <module>
  File "microdot.py", line 770, in run
OSError: -1
>Traceback (most recent call last):
  File "microdot.py", line 811, in dispatch_request
  File "microdot.py", line 257, in create
  File "microdot.py", line 318, in _safe_readline
OSError: [Errno 1] EPERM
Unhandled exception in thread started by <closure>
Traceback (most recent call last):
  File "microdot.py", line 36, in run
  File "microdot.py", line 860, in dispatch_request
  File "microdot.py", line 407, in write
OSError: [Errno 1] EPERM
```

Also, the socket lib seems missing the SOL_SOCKET, which means I cannot use the async implementation
`self.server.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)`
I think it is the same issue as in: [Issue](https://github.com/micropython/micropython/issues/8172#issuecomment-1013894522)

dpgeorge, 2022-02-18:
The Nina W10 driver is not yet fully compatible with `uasyncio`.  And the above error with the non-async version of microdot seems to imply that `readline()` is not working properly on the Nina W10.

iabdalkader, 2022-05-11:
I got this working with sync/asyncio, although I'm still debugging an issue with the sync version.

iabdalkader, 2022-05-11:
I think there might be an issue related to threading on rp2. The asyncio version of this works flawlessly, however the "sync" version serves only 1 request and then stops and drops to REPL, leaving a new socket open and bound to port `5000`, here's the first run (not server socket fd is 54, accepted socket fd is 55)

```
socket_send(55, 306)        # 55 is the accepted socket
close called via ioctl 55
socket_close(55)
GET / 200
>>>       # -> back to REPL.
```

If you do a softreboot, you can see the server socket (fd 54) closed via finaliser (stream_close/ioctl) and another socket with fd 56.
```
MPY: soft reboot
close called via ioctl 54
socket_close(54)
close called via ioctl 56
socket_close(56)
```

dpgeorge, 2022-05-26:
Should be fixed by be83bdf9ecec2395545357c75bfc4625631e27d5

closed 2022-05-26 by dpgeorge