micropython/micropython #8325: MicroPython OneWire protocol BUG?

beyonlo, 2022-02-18:
Hello!

I'm using the onewire CI DS28E05 (https://datasheets.maximintegrated.com/en/ds/DS28E05.pdf)  but Micropython `ow.scan()` show nothing.  That is a `onewire` chip, so it use/works with the `onewire protocol`, of course. Well, if it is `onewire protocol`, the CI address need to be shown on the ow.scan() right?

```
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.18.0', version='v1.18 on 2022-01-17', machine='ESP32 module (spiram) with ESP32')
>>> from machine import Pin
>>> import onewire
>>> ow = onewire.OneWire(Pin(25))
>>> ow.scan()
[]
>>>

```

Just to you know that `onewire` pin are working,  if I use another `onewire` chip, the DS18b20 CI, it shows the address on the `ow.scan()` as follow bellow:
```

>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.18.0', version='v1.18 on 2022-01-17', machine='ESP32 module (spiram) with ESP32')
>>> from machine import Pin
>>> import onewire
>>> ow = onewire.OneWire(Pin(25))
>>> ow.scan()
[bytearray(b'(\xa7I\xf0\x0c\x00\x00\x17')]
>>>
```
Any idea what is wrong? Is it a MicroPython OneWire protocol BUG or what I need to do to works the `ow.scan()`?

Thank you in advance.

peterhinch, 2022-02-18:
See [this forum thread](https://forum.micropython.org/viewtopic.php?f=16&t=11928&p=65221#p65221). Is it possible that your "Maxim" chip is a dodgy Chinese copy?

beyonlo, 2022-02-18:
Hi @peterhinch 

> See [this forum thread](https://forum.micropython.org/viewtopic.php?f=16&t=11928&p=65221#p65221). Is it possible that your "Maxim" chip is a dodgy Chinese copy?

wow, a fake chip is so bad. But I bought both MAXIM chips (the [DS28E05](https://br.mouser.com/ProductDetail/Maxim-Integrated/DS28E05R%2bT?qs=sGAEpiMZZMvKNHcrmI%252BvDQ2EyeHKWE%2F9WrTF7PS4xo8%3D) and [DS18b20](https://br.mouser.com/ProductDetail/Maxim-Integrated/DS18B20-SL+TR?qs=sGAEpiMZZMvC4tFtz7F%252BjoIerjBjnE4zDKdF%2FHZq9RI%3D)) directly from mouser.com site, I hope it is a trusted site.

* I tested two units of the DS18b20 (onewire) and all works on `ow.scan()`;
* I tested two units of the DS28E05 (onewire) nothing works - show nothing on `ow.scan()`.

Both kind `onewire` chips was bought from mouser.

I have two alternatives: 

1. Try to buy the DS28E05 (`onewire`) from another site and test it.
2. Find another `onewire` chip with `ID Unique Factory-Programmed` (from factory) that works on MicroPython onewire. Please, do you know one that you can suggest me? Yes, need to be `onewire`.

Thank you so much.

peterhinch, 2022-02-19:
I should point out I have no experience of these chips - I just recalled the forum thread.

As far as I know Mouser is a reputable supplier. But I was struck by the comment in the thread from a guy who runs an electronics distributor who said that keeping fakes out of the supply chain is very difficult.

beyonlo, 2022-02-19:
Hey, maybe there is light at the end of the tunnel :)

I see that in the DS28E05 datasheet show this : 

"
●	 Communicates with Host at Up to 76.9kbps
**(Overdrive Only)**
"

So I research on the internet something about that (**Overdrive**) and found this project on github where talk about a patch to works the `DS28E05` on `CircuitPython` **with overdrive speeds**: https://github.com/dragonlock2/MMv3  Look at the last part on the page with title **"Compiling CircuitPython"** There shows **"The DS28E05 1-Wire EEPROM operates at overdrive speeds only. Adding support involves compiling a patched version of CircuitPython."**

I do not understand very well as that works, but maybe is needed a patch to works in MicroPython too!

Could you please help me how/what to do that in MicroPython too?

Thank you in advance!

dpgeorge, 2022-02-21:
Looks like there's a patch for MicroPython as well: https://github.com/dragonlock2/MMv3/blob/main/micropython.patch

beyonlo, 2022-02-21:
Hey @dpgeorge 

> Looks like there's a patch for MicroPython as well: https://github.com/dragonlock2/MMv3/blob/main/micropython.patch

Oh, that's true. I didn't checked on the files :)

Please, could you to merge that patch in the mainstream to works by default in the next MicroPython release?  So, all `Overdrive` `OneWire` chips will works and all people will be benefited. Or that is a bad idea?

Thank you so much.

dpgeorge, 2022-02-21:
> Please, could you to merge that patch in the mainstream to works by default in the next MicroPython release?

That patch changes the timing for all onewire devices, so can't go in as it stands.  The timing values would need to be configurable so the driver works with all types.

beyonlo, 2022-02-22:
Hello @dpgeorge 

> That patch changes the timing for all onewire devices, so can't go in as it stands. The timing values would need to be configurable so the driver works with all types.

Understood. Could you please, change the the `onewire` driver to be configurable to works with all types of `onewire` chips?

Please, no not understand me wrong. I know that you have a lot of work to do on MicroPython and maybe this is not a priority for now. Just please, let me know at least if you have plans to do that in the future, because I already use others `onewire` chips that already working now - and, maybe,  they will stop to works if I use that patch. I would like so much to contribute making a PR, but I have not knowledge enough to do that.

I always used the ready compiled firmware from https://micropython.org/download/ Now I'm studying and learning how to compile/generate my own firmware to be possible apply that patch and test it on this `Overdrive` `onewire` chip that I have. 

Thank you in advance!

beyonlo, 2022-03-04:
@dpgeorge If you do not have intention to work in a configurable `onewire` driver for now, please, let me know if is possible to have two `onewire` drivers:  the original (`modonewire.c`) `onewire` and another (`modonewire_overdrive.c` - a copy of `onewire` drive with that patch applied) with name `onewire_overdrive`.

So in the pratical way (just thinking), will works in this way:

**Default OneWire:**
```
import onewire 
ow = onewire.OneWire(Pin(10))
ow.scan() #scan default onewire chips
```

**Overdrive OneWire:**

```
import onewire_overdrive 
owo = onewire_overdrive.OneWire(Pin(11))
owo.scan() #scan onewire Overdrive chips
```

Is possible to do that on with MicroPython? May be that can be a solution for me while has no official configurable `onewire` driver.  

Any other suggestion?

Thank you in advance.

beyonlo, 2022-03-08:
I tried to apply that [patch](https://github.com/dragonlock2/MMv3/blob/main/micropython.patch) for the MicroPython, but I have error when I try do build it:  `modonewire.c:76:5: error: implicit declaration of function 'busy_wait_us_32' [-Werror=implicit-function-declaration]`

**Details about the error is below:**

Applying the patch:

```
(viren) pi@pi:~/micropython/extmod$ patch < ~/Downloads/MMv3-main/micropython.patch 
patching file modonewire.c
Hunk #1 succeeded at 35 (offset 2 lines).
Hunk #2 succeeded at 61 (offset 2 lines).
Hunk #3 succeeded at 73 (offset 2 lines).
(viren) pi@pi:~/micropython/extmod$
(viren) pi@pi:~/micropython/extmod$ ls -la modonewire.c*
-rw-rw-r-- 1 pi pi 5857 mar  8 17:39 modonewire.c
-rw-rw-r-- 1 pi pi 5834 mar  8 16:41 modonewire.c.orig
(viren) pi@pi:~/micropython/extmod$
```
Error building:

```
(viren) pi@pi:~/micropython/ports/esp32$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/pi/micropython/ports/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/pi/micropython/ports/esp32"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/pi/micropython/ports/esp32/build/sdkconfig
Loading defaults file /home/pi/micropython/ports/esp32/build/sdkconfig.combined...
/home/pi/micropython/ports/esp32/build/sdkconfig.combined:8 CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE was replaced with CONFIG_COMPILER_OPTIMIZATION_SIZE
/tmp/confgen_tmpogzyhpk9:69 line was updated to CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
/tmp/confgen_tmpogzyhpk9:70 line was updated to CONFIG_BTDM_CTRL_MODE_BTDM=n
-- Found PythonInterp: /home/pi/micropython/ports/esp32/viren/bin/python (found version "3.8.10") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- App "micropython" version: v1.18-194-g507ad0332-dirty
-- Adding linker script /home/pi/micropython/ports/esp32/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/pi/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/pi/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
Found User C Module(s): 
-- Found Python3: /home/pi/micropython/ports/esp32/viren/bin/python3 (found version "3.8.10") found components: Interpreter 
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/pi/esp-idf/components/app_trace /home/pi/esp-idf/components/app_update /home/pi/esp-idf/components/asio /home/pi/esp-idf/components/bootloader /home/pi/esp-idf/components/bootloader_support /home/pi/esp-idf/components/bt /home/pi/esp-idf/components/cbor /home/pi/esp-idf/components/cmock /home/pi/esp-idf/components/coap /home/pi/esp-idf/components/console /home/pi/esp-idf/components/cxx /home/pi/esp-idf/components/driver /home/pi/esp-idf/components/efuse /home/pi/esp-idf/components/esp-tls /home/pi/esp-idf/components/esp32 /home/pi/esp-idf/components/esp_adc_cal [... cut ...]
/home/pi/micropython/extmod/modonewire.c:76:5: error: implicit declaration of function 'busy_wait_us_32' [-Werror=implicit-function-declaration]
     busy_wait_us_32(TIMING_WRITE1);
     ^~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[1269/1352] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/pi/micropython/extmod/moduheapq.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
(viren) pi@pi:~/micropython/ports/esp32$
```

Any idea/suggestion how to fix that?

Thank you very much.

jonnor, 2024-09-08:
A merge request for configurable timings in OneWire module would be welcomed.